baldram/ESP_VS1053_Library #1: To test the library with different hardware

baldram, 2017-07-05:
The issue was created to test the library with different boards and collect the result.

It will be published in one place here: [https://github.com/baldram/ESP_VS1053_Library/wiki](https://github.com/baldram/ESP_VS1053_Library/wiki).

**I ask you for help** to test the ESP_VS1053_Library and share the results.

baldram, 2017-07-05:
The library was tested with two ESP8266 based boards. 
I successfully run the example sketch and managed to play the test mp3 flow.

Below the hardware details:

- LoLin NodeMCU V3 (ESP-12F) + LC Technology VS1053 MP3 Codec board,
[image]

- RobotDyn WiFi D1 R2 (ESP-8266 EX) + LC Technology VS1053 MP3 Codec board.

baldram, 2017-07-09:
Below test results for ESP32 based board.  
A test mp3 from SimpleMp3Player example sketch was successfully played with the following `platformio.ini` settings.

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

lib_deps =
    baldram/ESP_VS1053_Library
```
- **Espressif ESP32 Core Board V2 / ESP32 DevKitC** + LC Technology VS1053 MP3 Codec board.

**Wiring details**:

| LC Technology VS1053 |  EspressifESP32 DevKitC |
|----------------------|-------------------------|
| SCK                  | GPIO18                  |
| MISO                 | GPIO19                  |
| MOSI                 | GPIO23                  |
| XRST                 | EN                      |
| CS                   | GPIO5                   |
| DCS                  | GPIO16                  |
| DREQ                 | GPIO4                   |
| 5V                   | 5V                      |
| GND                  | GND                     |

Aggebitter, 2017-12-20:
Noticed that every one is driving the board at 5V. Is the logic output 3.3V ?

I've just got the lib upp and running with a NANO32 and a Geetech VS1053 MP3 breakout board clone
(https://www.geeetech.com/vs1053-mp3-breakout-board-sd-card-slot-p-611.html)
But for the SD part I started with bypass the fixed AMS1117 3.3V regulator and drive the board at 3.3V

baldram, 2017-12-20:
@Aggebitter I'm glad you managed to run the lib. If you like it please do not forget to star the project :)

Yes, I successfully run those mp3 boards driving with 5V. 
If about the logic, LoLin NodeMCU V3 is 3.3V and works fine with VS1053.
I also have tested driving the VS1053 board differently. I connected 5V of LC Technology VS1053 board to 3V of LoLin NodeMCU V3 and it worked fine. I'm not sure it's only related to LC Technology one. Maybe it will work fine also with other manufacturer's boards.
For more details please see this comment from the thread above: https://github.com/baldram/ESP_VS1053_Library/issues/1#issuecomment-313163772 .

When you have the final version of your circuit up and running I would be appreciated if you share some photos or any other detailed description.

Aggebitter, 2017-12-22:
The VS1053b is a 3.3V logic chip but when driving the Geetech board at 3.3V the dropout voltage (1.1-1.3V) over the SDCard is to high (SD cards are very tricky on the voltage). I think that the logic output is at 3.3V when driving the board at 5V but I have not confirmed that. One benefit to drive the board direkt from 3.3V is that a single LiPo pack and charger can be used.  

I'm experimenting with Ogg recording so this was just a basic test to assure the Mp3 card is working

DeVampz, 2018-04-09:
where is XRST

DeVampz, 2018-04-10:
i have a problem about compiling esp32 with vs1053, what is CARDCS? And in the serial monitor, it shows nothing

DeVampz, 2018-04-11:
I have ESP Wroom 32 (aka ESP32 DevKitC) like in the pic 3 and i have installed enough toolchains, can you tell me more about platformIO.ini settings, i have installed libs, i still don't know about the Card Chip Select Pin, what port in the 2 modules that i have to connect with? 
[image]
[image]
[image]

DeVampz, 2018-04-11:
here is another code, i don't know how to add vs1053.h into the board
[image]

baldram, 2018-05-04:
@DeVampz Sorry for late reply. I'm in the middle of flat renovation and I have a have little free time recently. 

Have you managed with your issue already? 
You've posted two questions one by one. When look at the screen of the first source code I see that it's not connected with Baldram's ESP_VS1053_Library but Adafruit_VS1053 library so I would look for help in some Adafruit related forum. Btw. when look on the message "Can't find VS1053..." I think you have an issue with wiring of your board since your Wroom 32 doesn't have to be the same as Feather ESP32. So even single pin may be connected wrong and you will see message like this. It may be even GND pin. But no idea what exactly, so please ask people from Adafruit for better explanation. If about Card Chip, (as mentioned previously) my ESP_VS1053_Library has nothing to do with SD card. This library drives VS1053 audio only. 

The code from the second screenshot is correct. I see an example for ESP_VS1053_Library. You ask how to add vs1053.h. It looks like you don't use PlatformIO platform which resolves dependencies for you automatically. The ESP_VS1053_Library (vs1053.h) is a library for mentioned PlatformIO platform. It's a great tool and I urge you to use this amazing platform. You will find more information [here](https://platformio.org/get-started): https://platformio.org/get-started. 

> ... tell me more about platformIO.ini settings

In my previous answer I pointed you to place where you have a [complete content of the platformio.ini file](https://github.com/baldram/ESP_VS1053_Library/issues/1#issuecomment-313907348).  As you use ESP32 DevKitC everything is exactly the same. In case of other board you need to modify it [properly](http://docs.platformio.org/en/latest/platforms/espressif32.html#boards) by providing a [proper](http://docs.platformio.org/en/latest/platforms/espressif32.html#boards) `board` and `platform` only. That's all. Easy. 

In case of other issues with general PlatformIO usage please don't hesitate to ask a question here: https://community.platformio.org/. 

This thread is more for documenting tested boards. So if you already managed to run examples for ESP_VS1053_Library with your ESP Wroom 32 (aka ESP32 DevKitC), please leave an info here. It will be very appreciated. It should work as it's compatible with already tested ESP32 Core Board V2.
 
But it would be nice to test as much boards as possible. In this thread I ask community for help ;-)

PS: Please do not forget to appreciate my work and leave a star if you like this Github repo.

Boredot, 2018-07-07:
Can't you help me with my problem? I know, this project is old but I already want to ask you a question. Usually I use arduino IDE, but you wrote that this library is mentioned for Platformio, so i tried it. I have this issue and don't know what can I do here.
[image]

baldram, 2018-07-09:
@DeVampz This issue is named "To test the library with different hardware". I ask people to test the library and share the result here. I would like to keep this topic clean with only hardware testing results. 

In case of problems like the one you mentioned, I kindly ask you to create a new issue.
Would you please create a new issue? Please put the question and screenshot there. Then we can try to discuss it and I will try to give you some hints.

LucasMaciuga, 2020-01-17:
The webradio example works nice with the "New ESP8266 Programmer Socket Adapter For ESPRESSIF". Pinout is the same as for "LoLin NodeMCU V3" - just use the correct I/O ports (I used this as a mapping reference: https://www.theengineeringprojects.com/wp-content/uploads/2018/10/Introduction-to-NodeMCU-V3.png).

[image]

Nice project - nice work.

Lucas

baldram, 2020-01-18:
@LucasMaciuga Thank you for sharing your set-up. I'm happy to see it works well. 
I have one question. While this is an adapter, you probably have the possibility to put there different ESP modules (from specification I see it supports: ESP-12S/12F/12E/07S/07). Which module do you use in your set up from the picture?

LucasMaciuga, 2020-01-18:
@baldram for the setup in the picture I used an ESP-12F.

LucasMaciuga, 2020-01-18:
I also attached the VS1053 to a 'blank' ESP-12F. Works great.
Take note that I used a LM1117T-33 to bring a 9V battery to 3,3 V - seems to be enough for the VS1053.

[image]

baldram, 2020-01-19:
@LucasMaciuga That's interesting. I wonder how long such battery can drive ESP with VS1053, e.g. for WebRadio? WiFi needs a lot of energy.

baldram, 2020-01-31:
@LucasMaciuga My question above could be overlooked. Did you have a chance to observe or measure how long such a battery can drive the ESP?

ewbarnes53, 2020-04-07:
I am using your library and examples, using Adafruit Feathers and Wings.  I have the Huzzah32 ESP32 Feather (PID = 3619) with the Music Maker Wing (PID 3357).  I have both the WebRadio and MP3PlayerDemo working.

The problem I have is the audio quality on the WebRadio is very poor.  It sounds very good most of the time, but then there is occasional popping, clicking, and chirping on the audio.

The problem is with all of the stations I have been listening to.  For example:
`//  1940 UK Radio
const char *host = "149.255.59.162";
const char *path = "/1";
int httpPort = 8062;`

If I use one of the Adafruit examples to play music files, they sound fine.  Actually, they sound quite good.  I can play MP3s, WMAs, and WAVs.

I wonder if you have any suggestions about what might be the cause of all of the noise on the audio.  In the for what it is worth, everything in my system is running at 3V.

philippedc, 2020-04-07:
@ewbarnes53 can test with HTTP1.0 instead of HTTP1.1 and share the result ?

baldram, 2020-04-07:
> @ewbarnes53 can you test with HTTP1.0 instead of HTTP1.1 and share the result?

@philippedc @ewbarnes53 This is a good idea. 
To keep the discussion consistent, I would suggest sharing the result directly in issue #52 (related to clicking). 
Ok?

> I am using your library and examples, using Adafruit Feathers and Wings. [...]. 
> I have both the WebRadio and MP3PlayerDemo working.

Thank you for using the library and sharing results. 

> The problem I have is the audio quality on the WebRadio is very poor. It sounds very good most of the time

Please note that the attached example is a very simple code. Kind of "hello world" for web radio simple program. So the problems are related to implementation. With such a simple implementation like one from WebRadioDemo it's hard to expect a lot. 
If about a more complex example, it's the "Edzelf/Esp-radio" project itself. This is the one from which I extracted the VS1053 driver code. The Edzelf's code does a lot and deals with buffering and so on to handle different issues.

> there is occasional popping, clicking, and chirping on the audio.
> [...] I wonder if you have any suggestions about what might be the cause of all of the noise on the audio. 

As @philippedc already mentioned, there is the other thread where this problem is discussed.
In fact, I see three threads I would suggest looking at for the glitches/popping/clicking issue:
* An earlier discussion #44. Further two other issues on the same topic were created (ones below).
* The mentioned #52 is the one where @philippedc suggested using HTTP1.0 as a workaround.
* The long discussion regarding buffering #47 and example codes extending the radio example.
The clicking problem might be related to a chunked stream. Out of the mentioned discussion @wmarkow created an (ongoing) PR with [a new WebRadioChunkedDemo.ino example](https://github.com/baldram/ESP_VS1053_Library/pull/48/files). 
Would you like to test this code with your stations?

Would you please share your thoughts directly on related issues?

> If I use one of the Adafruit examples to play music files, they sound fine. Actually, they sound quite good. I can play MP3s, WMAs, and WAVs.

What do you mean Adafruit examples? Which ones? Are they also for WebRadio? Or they just play sound files like mp3 from the board memory or SD card? 
Then I guess you should have similar results with MP3PlayerDemo, aren't they?

ewbarnes53, 2020-04-08:
Thank you for the responses.  I will look at the other threads.  I am new to Github, so I am still trying to learn my way around.

> @ewbarnes53 can test with HTTP1.0 instead of HTTP1.1 and share the result ?

I did modify my code to use HTTP1.0.  I did not hear any change.  I am not having problems with not being able to connect, or being disconnected.  I have been connected to the station I mentioned all afternoon, and it is still playing.

I did run the MP3PlayerDemo.  It works, but is such a short tone, that I couldn't tell much.  I listened to the tone you recorded, and they sounded the same (to my very old ears).

Adafruit has also created a VS1053 library.  They have a Fileplayer class, that plays files directly from the SD card.  Their demo sketch also has some problems, but the problems relate to not handling interrupts correctly when the ESP32 is used, which seems to be an issue they know about.

I will try to find and figure out the chunked demo, and then let you know what I have found.

And again, I will say that I am new to GitHub, and am more of a Hacker than an experienced programmer.  I want to thank you guys for releasing your libraries and the sample code.  I have not had much luck with figuring out how to use libraries, unless sample code using the library is included.

Givy, 2021-01-13:
Hi,
This is an interesting project. So, I bought an Adafruit VS1053B (Codec + MicroSD Breakout - MP3/WAV/MIDI/OGG Play + Record - v4) and a Heltec_ESP32 WIFI LoRa 32(V2).  I have not had much luck in getting it working yet.  What could be the Pin connection differences compared to the ESP32 that you have shown?  
Also, would it be possible to display the Radio Station Name/Number/Country on the OLED that comes on this ESP32?  Should that be stored on an external EEPROM etc?
Thank you for your input.

baldram, 2021-01-14:
Hi @Givy thank you for using the library. 
I had no possibility to test Heltec_ESP32 myself but if you get luck it would be great if you share your results (I've just added [Discussions](https://github.com/baldram/ESP_VS1053_Library/discussions) tab to the project). I don't expect any specific differences. Here is [DevKitC](https://components101.com/sites/default/files/component_pin/ESP32-Pinout.png) and here is [yours Heltec](https://nettigo.pl/system/images/3656/original.jpg?1588067482). Looking at the schema I see you have also all those mentioned GPIO18, GPIO19, GPIO23 and so on and so forth. Maybe different side, different order, but they are there. I would try the same wiring but using the correct places for pins on your board.

> Also, would it be possible to display the Radio Station ... that comes on this ESP32?

I believe it would. 😏 But it's not in the scope of this audio driver library 😄 
I don't know here is something and it looks like [a library](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/src/oled/API.md) from manufacturer of your board. 
The ESP_VS1053_Library is just a VS1053 driver and even has nothing to Web Radio. The "radio", "mp3 file play" or "beep sound" are just example use cases, that users might try to implement using this lib.

> Should that be stored on an external EEPROM etc?

Not sure I understand your question. The display handling will be simply a part of your arduino project. 
I extracted the audio driver from the [parent project](https://github.com/Edzelf/Esp-radio) (from where this fork is made).
The parent project has a one file app where is everything. The audio, display handling, wifi, mqtt and more. All in one file.
Now, [somewhere here](https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino#L336) starts the VS1053 library. That's it. So let's say, you see an example of the complete application using this library.

Good luck! Please do not forget to click "star". 
Please also share your results when you get your board up and running. Maybe a photo of your setup.

Givy, 2021-01-14:
Hi,
Thank you so much for responding.  I hope I didn't go too much overboard with my questions 😱.  Thank you for the additional information.  Also, I was referring to the additional information such as "Station Name/Country/etc" to be stored on the EEPROM, or if it can scan the web actively and find that information and display.   I'll work on it and will let you guys know if I was successful.  😀

Thanks.

Givi

baldram, 2021-01-15:
Ok, now I understand EPROM related question. So for "Station Name/Country/etc" I would use the network. The ESP board has easy WiFi access, so why not benefit from this? The cloud storage sounds so up to date, but you can use some local LAN server (local cloud? 😅). Your use case ("internet radio") is about connecting the network anyway, so everything would be network oriented. Sounds good to me.

Good luck.

strider222, 2021-03-05:
Hard to tell what the board is exactly. I think it is a ESP32 30 pin dev board. Above you show a 36 pin board. On the back of the board is says ESP32s and omits the words dev board so I'm not sure I got what I ordered. The pinout is different as per below. 
The VS1053 board looks identical to the one shown by Lucas on Jan 17, 2020.

I had some runtime code errors but once fixed it has been working for 24 hours.

Wiring:
  --------------------------------
  | VS1053  | ESP8266 |  ESP32-30pin   | 
  -----------------------------------
  |   SCK   |   D5     |   IO18 |
  |   MISO  |   D6    |   IO19 |
  |   MOSI  |   D7    |   IO23  |
  |   XRST  |   RST |   EN     |
  |   CS      |   D1    |   D32 XCS
  |   DCS   |   D0    |   D33 XDCS
  |   DREQ |   D3   |   D35    |
  |   5V       |   5V    |   5V     |
  |   GND   |   GND |   GND  |
  --------------------------------

|----vs1053 board--------ESP32-30pin  ------
|   --------------
|   |5v     | 5v     |            VIN   VIN
|   |dgnd | miso |            GND  D19
|   |mosi | sck   |    ==     D23   D18
|   |dreq  | xrst  |             D35   EN
|   |xcs    | xdcs|             D32   D33
|   --------------
|                    Jack     
|                      X
|                      X         
|------------------X------------

I ordered Geekcreit® ESP32 WiFi+bluetooth Development Board which on banggood's website shows ESP32 DEVKITV1 and mine shows ESP32s 

Main page for order
`https://m.banggood.com/Geekcreit-ESP32-WiFi+bluetooth-Development-Board-Ultra-Low-Power-Consumption-Dual-Cores-Pins-Unsoldered-p-1214159.html?akmClientCountry=CA&utm_design=41&utm_source=emarsys&utm_medium=Neworder171109&utm_campaign=trigger-order&utm_content=winna&sc_src=email_2675773&sc_eh=b45feb052732c2c31&sc_llid=17049149&sc_lid=105229698&sc_uid=jcfXFTXc6i`

More product details
`https://m.banggood.com/Geekcreit-ESP32-WiFi+bluetooth-Development-Board-Ultra-Low-Power-Consumption-Dual-Cores-Pins-Unsoldered-p-1214159.html?akmClientCountry=CA&utm_design=41&utm_source=emarsys&utm_medium=Neworder171109&utm_campaign=trigger-order&utm_content=winna&sc_src=email_2675773&sc_eh=b45feb052732c2c31&sc_llid=17049149&sc_lid=105229698&sc_uid=jcfXFTXc6i`

strider222, 2021-03-05:
The above based on the code from the examples will not play any BBC station although the link I think that is generated plays in my browser. The client.connect() returns success but client.connected is false.
"BBC 6 Music",              "bbcmedia.ic.llnwd.net",    "/stream/bbcmedia_6music_mf_p", 80,   
"BBC World Service",        "bbcwssc.ic.llnwd.net",     "/stream/bbcwssc_mp1_ws-eieuk", 80,
 
This station produces a return code of zero on connection but has been known to play occasionally.
      "Comet radio",              "comet.shoutca.st",  "/1",                                  8563, 
I think this means that the VS1053 plays AAC files.
      "Scala Radio",              "stream-mz.planetradio.co.uk", "/scalahigh.aac",            80,

Playing AAC also means that player.switchToMp3Mode() is actually switch to auto-decode mode as it plays AAC. Or perhaps it is switchOutOfMidiMode(); I'm wondering if the BBC stations are streaming an unrecognized file format (or I got a broken VS1053)

I would love a link on how to identify the URLs to use from Linux when I connect to a local radio station on my browser.

Dr-Dawg, 2021-03-08:
Hi, I just tested both BBC stations (with a modified code using ringbuffer), both work fine, no redirection or anything.
MPEG 1 Layer 3, Bitrate 128, Sample Frequency 44100

As a hint, in the header  the icy- description says "BBC radio will stop working on this device soon", but this could be an issue of me defining an user agent.

Concerning the AAC format, as far as I remember, the blue LC Technology boards contain a VS1003 chip which, unlike the VS1053 on the green boards, do not play AAC streams (not sure anyway).

Dr-Dawg, 2021-08-13:
[image]

Hi all, I just wanted to share my current progress with you..
Turned a 1 Euro clock into a web radio with LCARS design, Web Server, Gesture recognition (PAJ7620U2) and gyroscope (MPU6050).
All started with the WebRadioDemo and Arduino IDE,  but if I knew in advance the time it would take me and the functionality I would add, I would have probably started with Edzelfs ESP32-Radio. The only thing I "borrowed" from his code for now is the showstreamtitle function..
Switched from ESP8266 NodeMCU v3 to NodeMCU ESP32 and ESP-32 Dev Kit C V4. I learned a lot, from reading MP3 frame sync to using both VSPI and HSPI on the ESP32 and putting a condensator between ground and enable for not having to push the boot button for each upload.
Thanks again to all of you for all the helpful contributions posted here!

episource, 2022-03-22:
Building on PR #90 the lib is compatible with a [Raspberry PI Pico (RP2040)](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html) as well. This has been tested using the [arduino-pico core by earlephilhower](https://github.com/earlephilhower/arduino-pico) via platformio and the [VS1053B breakout board made by Adafruit](https://learn.adafruit.com/adafruit-vs1053-mp3-aac-ogg-midi-wav-play-and-record-codec-tutorial/overview).

[image]

episource, 2022-05-26:
Here's the follow up link to the final setup where I'm using this lib (with PRs #98 and #99 merged): https://github.com/episource/picopdio

jenschr, 2023-05-04:
Thanks for making this project! To use this library with Adafruit's VS1053B breakout, be sure to reset the breakout as in this example:
https://gist.github.com/jenschr/dacf60d2c9d2bd4efee85768cd5eab82

Tested with an ESP32-S3 & the Adafruit VS1053 Codec + MicroSD Breakout (https://www.adafruit.com/product/1381)