baldram/ESP_VS1053_Library #2: Examples to stream a mp3 from web server

Uksa007, 2017-10-22:
Hi,

Are there any examples of how to stream a single mp3 from a web server available?    

I note that playChunk seems to be blocking, is there any option in the library that use interrupts to make it non-blocking? I'm running a web server on the ESP8266 and so blocking is not so good.

void playChunk(uint8_t *data, size_t len);  // Play a chunk of data.  Copies the data to
                                                // the chip.  Blocks until complete.

Thanks.

baldram, 2017-11-08:
Hi Uksa, sorry for inactivity. I'm a bit busy recently. I will have a look to your question soon.
Regards, Marcin

baldram, 2017-11-22:
ping @Uksa007

Uksa007, 2017-11-22:
Hi,

I had a look at ESP32-Radio, its a complex beast, kind of hard to pinpoint exactly how it streams. 
I also don't have much luck getting it stream without pausing, maybe the buffer is too small.

I ended up writing my streaming code, which mostly works, just working on making it non blocking.

baldram, 2017-11-22:
Thanks for your reply. I'm happy you've managed to solve the issue. 
I will close this ticket then for now. 

In the near future I have a plan to provide some more examples regarding different things.
Also the community is very welcome to add some examples using this library.

closed 2017-11-22 by baldram