baldram/ESP_VS1053_Library #4: mp3 stream for webradio

zhooley, 2018-02-01:
Hi!

The example is working, but how to make a webradio?

Thanks

baldram, 2018-02-01:
Hi, thank you for the feedback!

The library delivers basic functionality to be used for different purposes. One of them may be webradio.
A good example of webradio is a parent project from which I extracted the basic functionality to deal with VS1053 board. 

The parent project I mentioned is https://github.com/Edzelf/Esp-radio as well as https://github.com/Edzelf/ESP32-Radio. Unfortunately both are not made of components, just everything mixed togheter in one place. 

But you may find there a way how to receive a radio stream. It would be great to create a simple example for ESP_VS1053_Library showing how to get radio functionality. If you find a time and manage to do it, please feel invited to publish your change to the repository in "examples" folder.

baldram, 2018-08-01:
Hi @zhooley . Please find a webRadio example by @vincegellar using this ESP_VS1053_Library.
https://github.com/vincegellar/Simple-Radio-Node 

The issue is about to be closed then.
Probably in the future I will add my own simple radio example to the examples folder or I will ask Vince for a pull-request.

closed 2018-08-01 by baldram

vincegellar, 2018-08-01:
Hello @baldram !

Thanks for the mention! I've prepared a commit demonstrating the webstream handling. I've basically removed the volume adjustment parts, and added the code to the examples folder. Could you grant push access for the commit?

Thanks!

baldram, 2018-08-01:
Hello @vincegellar ! Thank you for very quick response! Would you please add this change via a pull-request? It works this way for most Github projects. 
PS: Maybe I will have some suggestion regarding Platformio related things while doing a code review. 

Thanks!

vincegellar, 2018-08-01:
@baldram I feel a bit dumb now, I'm did not realize I could use a forked repo for pull requests. I'm pretty new to collaborating on github. 
I've created the pull-request. Looking forward to any insights.

baldram, 2018-08-02:
@vincegellar Thank you very much. If about pull requests, it's rather my fault. I haven't added collaboration guideline to the project documentation. I will fix it soon. Thank you again and I go to have a look at your pull request...