baldram/ESP_VS1053_Library #7: Cannot compile on Arduino IDE

palsbo, 2018-06-21:
No matter what I do I get "avr/pgmspace.h: No such file or directory" error.
What to do?

baldram, 2018-06-21:
Could you please tell more what you do?
Can you provide steps you performed to compile this stuff? 

As you know this is PlatformIO library. How do you use PlatformIO to build the project?
Do you call commands directly from command line? 
Or maybe you build it and upload firmware to the board from your IDE?
(If yes, what exactly IDE do you use)?

baldram, 2018-06-27:
Hi @palsbo 
Have you seen my answer?

palsbo, 2018-06-30:
Hi
Thank you for reply and sory for my slov reply.
I was trying to use the library with Arduino IDE . But I realize that it
will take some significant chages.

baldram, 2018-06-30:
@palsbo In fact this library is designed to be used with **arduino Core for ESP8266/ESP32**.
The only important thing is that it is also designed as a plugin for PlatformIO (see more here: https://platformio.org/get-started). Please see platformio as dependencies manager. It wil l resolve all necessary dependencies for you. Then you can develop your project using your favorite IDE like: Atom, Visual Studio Code, CLion, Eclipse and more. Even VIM if you like. See here: http://docs.platformio.org/en/latest/ide.html and here: https://platformio.org/platformio-ide . 

But still the most important thing if you use ESP_VS1053_Library is to build the project with PlatformIO environment which is great and automatize lot of things for you. I personally use CLion or VSCode. Even I have IDE integration, I prefer to run commands myself from CLI. 

I will update the documentation to emphasize (somewhere in the beginning) that it is PlatformIO library. Then I will close the ticket.

baldram, 2018-07-07:
The issue to be closed now. 
I updated README.md with PlatformIO related information.

closed 2018-07-07 by baldram