baldram/ESP_VS1053_Library #10: ArduinoLog

Boredot, 2018-07-09:
Can't you help me with my problem? I know, this project is old but I already want to ask you a question. Usually I use arduino IDE, but you wrote that this library is mentioned for Platformio, so i tried it. I have this issue and don't know what can I do here.
[image]

baldram, 2018-07-09:
This is PlatformIO related question. I googled it for you. Have you tried? ;-)

Please find the platform related FAQ and answer here:
http://docs.platformio.org/en/latest/faq.html#unicodewarning-unicode-equal-comparison-failed

If it doesn't help or in case of further PlatformIO related issues I think it's a good place to find an answer. 
PlatformIO has a great forum. You can ask your question there or simply search for similar issues already asked by other users: https://community.platformio.org/ .

Please let me know about the result and I will close this issue.

Boredot, 2018-07-15:
Sorry foe the so late reply, i was busy these days. Well, after just redownloading the library but in Atom(platformio) everything worked. BUt another thing happened. Project uploads, but doesn't work. I don't think i wired something wrong and already checked it, so can't you say me, what can it be?(won't upload a screenshot because everything is good at the command string)

baldram, 2018-07-15:
Thanks for feedback. Without seeing your setup I'm not able to say you what is wrong.

Would you please provide then some details:
1) what is the project you uploaded? Was it example, the one from screenshot above? 
So [this one](https://github.com/baldram/ESP_VS1053_Library/blob/master/examples/SimpleMp3PlayerWithDebug/SimpleMp3PlayerWithDebug.ino)? 
Have you changed anything in the original example?
2) What is your ESP bard? The model, manufacturer, etc? 
3) What is your VS1053 board and manufacturer? 
4) Would you please provide information about wiring, both ESP board and VS1053 board, like I did in my description here: https://github.com/baldram/ESP_VS1053_Library/issues/1 (see "wiring details") for 3 different boards. For each board combination it's properly adjusted. I'm interested how does it look in your case.
5) If it's not an issue, it would be good if you upload the picture.

Boredot, 2018-07-17:
I uploaded an example and didn't change anything. Wiring is right like at the example because i have wemos d1 based on esp8266
[image]My mp3 decoder is quite common too. It's vs 1003/1053 module so I think it can work with this library.

Boredot, 2018-07-17:
Really don't realise where is the problem. Mb i need to check vs1053 with board like arduino uno just to know that it works?

baldram, 2018-07-17:
Thank you for reply and picture. 
Unfortunately I'm not able to see wiring details from the photo.
As you mentioned you have wiring like in the RobotDyn D1 example here:  https://github.com/baldram/ESP_VS1053_Library/issues/1#wemos-d1 .

But when I look at your picture I'm not able to see each side of the board. It's hard to guess what is there.
Could you simply please write down which pin of ESP is connected to which pin of VS1053 like I did [here](https://github.com/baldram/ESP_VS1053_Library#example-wiring)? Of course please use full pin description from your board.

For instance, I'm not sure, but when I compare your photo and mine of [RobotDyn D1](https://github.com/baldram/ESP_VS1053_Library/issues/1#wemos-d1) I see the part close to power may be connected in a different way. 
When I was experimenting, even a small thing in wiring was a problem.

> Mb i need to check vs1053 with board like arduino uno just to know that it works?
Of course if you have a possibility to check it with other library (this one doesn't support Arduino Uno) it would be helpful to know that VS1053 board itself is ok.

Boredot, 2018-07-18:
[image]
[image]

VS1053 | ESP8266
-- | --
SCLK | D5
MISO | D6
MOSI | D7
XRST | RESET
XCS | D1
XDCS | D0
DREQ | D3
5V | 5V
GND | GND

Right now I really started thinking that something is wrong with wiring. My vs1053 board has got another pins, and I'm not really sure in connecting, most of all at the SCK(SCLK on my board I think. 

Thank you for your help and answers

baldram, 2018-07-18:
Serial Clock (SCK or SCLK) should be fine. 
I was not sure how it looks if about RESET connection. I had myself some issue with incorrectly connected reset and after fixing everything has worked. 
Currently I'm in the middle with flat renovation. I would have to find and unpack this stuff from one of the boxes (not sure where it is). I would connect everything again with Wemos D1 compatible board. Maybe I will find it and also find a bit of time. 

In the meantime please check one thing. Here is previous version of `helloMp3.h` sound file: 
https://github.com/baldram/ESP_VS1053_Library/blob/bfa28d828c48ae58207493e02f5e18ef09b4c67a/examples/SimpleMp3Player/helloMp3.h
Someone has reported that "tick-tock" test sound is not good enough and sounds like a noise. 
So maybe it's a good idea to check it with the old sound.

baldram, 2018-07-20:
Hi @Boredot , I managed to setup everything again.
I've also tested a new [sound file](https://github.com/baldram/ESP_VS1053_Library/issues/5#issuecomment-377343083) which will be uploaded to the project soon. Below some information what I do. Quite helpful is to watch te Serial Monitor to see some debug messages. Please find the steps below (command I run from the command line) and the output I get.

### 1. Build and upload

```
$ platformio run --target upload
******************************************************************************************************************************************************************
Obsolete PIO Core v3.4.0 is used (previous was 3.4.1a6)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
******************************************************************************************************************************************************************
[Fri Jul 20 05:37:15 2018] Processing nodemcuv2 (platform: espressif8266; lib_deps: baldram/ESP_VS1053_Library; board: nodemcuv2; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 26 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ArduinoLog> v1.0.1
|-- <ESP_VS1053_Library> v1.0.6
|   |-- <ArduinoLog> v1.0.1
|   |-- <SPI> v1.0
Compiling .pioenvs/nodemcuv2/src/main.o
Looking for upload port...

Warning! Please install `99-platformio-udev.rules` and check that your board's PID and VID are listed in the rules.
https://raw.githubusercontent.com/platformio/platformio/develop/scripts/99-platformio-udev.rules
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/nodemcuv2/firmware.bin
Uploading 243072 bytes from .pioenvs/nodemcuv2/firmware.bin to flash at 0x00000000
................................................................................ [ 33% ]
................................................................................ [ 67% ]
..............................................................................   [ 100% ]
================================================================== [SUCCESS] Took 26.32 seconds ==================================================================

```
### 2. Click reset button (your ESP device) and use phones to hear the sound
Directly after successful firmware upload it did't work (no sound). When I clicked reset button it started to play immediately.

### 3. Check serial monitor to see what happens (the command below executed while device is playing a test sound)

```
$ platformio device monitor
******************************************************************************************************************************************************************
Obsolete PIO Core v3.4.0 is used (previous was 3.4.1a6)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
******************************************************************************************************************************************************************
--- Miniterm on /dev/ttyUSB0  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
N: Playing sound... N: Playing sound... N: Playing sound... N: Playing sound... N: Playing sound... N: Playing sound... 
```
### 4) Another experiment with serial monitor to see what happens when reset button is clicked.
```
$ platformio device monitor
******************************************************************************************************************************************************************
Obsolete PIO Core v3.4.0 is used (previous was 3.4.1a6)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
******************************************************************************************************************************************************************
--- Miniterm on /dev/ttyUSB0  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␎HpI<�bFS␞�H<�␟��N: 
Hello VS1053!N: Reset VS1053...
N: End reset VS1053...
N: Slow SPI,Testing VS1053 read/write registers...
N: Fast SPI, Testing VS1053 read/write registers again...
N: endFillByte is 0x0
N: Playing sound... N: Playing sound... 
--- exit ---
```
I wonder how the serial monitor output looks like in your case.

Later in the evening I will write you my wiring used this time and photo, but I connected everything like in the README.md description.

Boredot, 2018-07-20:
Ok, I tried to open Serial Monitor and found one interesting thing. Don't know what can it be?(I think there is something with the sound, or I'm thinking wrong)
[image]

P.S. Every other baudrate gave only strange sympols

baldram, 2018-07-20:
As it's 9600 it should set to 9600.
I run it from CLI as you see from my previous answer with command `platformio device monitor`.

baldram, 2018-07-20:
I see from your screenshot you set it to 74880 baud.
If your code has it like this:
`Serial.begin(9600);`
You should also use the same value for your Serial Monitor.
As it's 9600 it should set to 9600.

baldram, 2018-07-20:
The wiring for RobotDyn WiFi D1 R2.

VS1053 | ESP8266
-- | --
SCK | SCK/D5
MISO | MISO/D6
MOSI | MOSI/D7
XRST | RST
XCS | SCL/D1
XDCS | D0
DREQ | D3
5V | 5V
GND | GND

[image]

baldram, 2018-07-20:
A new example sound is now available here: https://github.com/baldram/ESP_VS1053_Library/blob/master/examples/SimpleMp3Player/helloMp3.h

Boredot, 2018-07-21:
Well, line you said about I already had at the project an checked now(switchToMP3) and I already putted volume to 100. I added a Serial.begin line and alright have only strange symbols on the 9600 baud
[image]
(Here you can see what I have after pressing a RST button) Think today i'll try new wiring scheme and tell you tomorrow how it works. Thank you again for everything you are doing to me

baldram, 2018-07-21:
Hmm... that's strange. It looks like something is wrong then and VS1053 was not initialized correctly. 
But it's strange as at least something what comes from example sketch should be visible. So that line:
`Log.notice("Hello VS1053!\n");` ... and I even don't see this "Hello..." text. So somehow it behaves like Serial Monitor doesn't show anything what is expected.
However in my case it looks like below. 
Please click "star" if you appreciate my involvement. Hopefully you'll have some findings. Good luck.

As this project is coming back in the meantime I will implement soon some very simple web-radio example to satisfy ticket #4.

[image]

Boredot, 2018-07-22:
Can't you tell me where I shall put this Log.notice line? I gonna also buy new cables(don't have enough at home and tell you if something gonna be better. 

Sorry, I leave my house for about 5 days tomorow, so I'll tell something new only later. I really like this project and hope you'll earn a lot of stars on it.

baldram, 2018-07-22:
It is already there. Please see Log.notice in Line 73 of this example:
https://github.com/baldram/ESP_VS1053_Library/blob/master/examples/SimpleMp3PlayerWithDebug/SimpleMp3PlayerWithDebug.ino

Interesting thing is that even output related to this is not visible on your screenshot. So it looks like Serial Monitor doesn't show everything. Later each 3 seconds "Playing sound... " should be shown what comes from line 80. So maybe it's output of something else. 
You can also try to change line 80 to `Serial.begin(115200);` and then start monitoring using 115200 baud rate.
Good luck.

baldram, 2018-07-28:
I've just provided a new change to the VS1053 library. The ArduinoLog dependency has been removed completly.
But it can be still used in own projects or in example. However the attached example also has been cleaned up and it doesn't use this external logging library now.

The documentation is updated properly.
Now build flags should be added properly (in `platformio.ini` file) to see debug output.
https://github.com/baldram/ESP_VS1053_Library#logging--debugging

**In other news, I have a hint for you**. Once testing the library after the change (with both ESP32 and ESP8266) I've discovered an issue you may have. There was **no contact for one of the cables** (even wiring was correct in general). **Thus I had no sound** (like you?). Even you have everything done well or you solder cables for instance, it's also possible that one of cables is broken or so... Then **just such a hint** (a solution in my case).

baldram, 2018-08-22:
@Boredot knock, knock... ;-)

baldram, 2018-10-16:
It looks like the problem was due to hardware issues.
Time to close the ticket. A lot of helpful information has been provided already. 
In case of further questions we can still reopen it...

closed 2018-10-16 by baldram