baldram/ESP_VS1053_Library #18: How to convert MP3 file to C Header file

williamesp2015, 2018-07-29:
VEry nice project. I could play sound on a Arduino MP3 shield board with TF_card. I need a converter to use in my program.
Thank you

baldram, 2018-07-29:
Hi @williamesp2015 .
Thank you! If you like the project please do not forget to click the star. Thanks :)

If about the audio file conversion.
I convert the sound files to C using Linux shell command:
`hexdump -v -e '16/1 "%4u,"' -e '"\n"' Click-sound.mp3 > click-sound2.c`

It produced something like this: https://github.com/baldram/ESP_VS1053_Library/blob/46cbc6435b1bcf19ee52b0f61a8a9d0cb925f3d2/examples/SimpleMp3Player/helloMp3.h

Recently I found something what produces nicer output. 
[Current  version](https://github.com/baldram/ESP_VS1053_Library/blob/master/examples/Mp3PlayerDemo/SampleMp3.h) was converted this way:
`xxd -i ding.wav sound.c`
and it produces something like this from wav:
```
unsigned char ding_wav[] = {
  0x52, 0x49, 0x46, 0x46, 0xb0, 0x19, 0x00, 0x00, 0x57, 0x41, 0x56, 0x45,
...
```
and from mp3:
```
unsigned char ding_mp3[] = {
  0xff, 0xe3, 0x80, 0xc4, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
...
```
using this software:
`xxd V1.10 27oct98 by Juergen Weigert`

Good luck!
Regards,
Marcin

williamesp2015, 2018-07-29:
Thanks @baldram, Very nice but practically it is difficult to convert and I am thinking to play MP3 file on SD memory card. I couldn't find compatible example with your code using SD card.

baldram, 2018-07-29:
Hi, I provided you various possibilities for doing the same. But all you need to do is just to call from command line:
```
xxd -i your-sound.mp3 ready-to-use-header.c
```
Can't be easier I think.

The xxd util is a part of Vim editor. So it's not only for Linux users. If you install Vim, you can even call this command under Windows.  Please find related thread [here](https://superuser.com/questions/497953/convert-hex-dump-of-file-to-binary-program-file-on-windows#638850). 

> I couldn't find compatible example with your code using SD card.

SD card is another peace of hardware. Of course VS1053 driver can be combined with that or anything else. So you can go this direction of course. Unfortunately I don't have any tested (myself) example to provide you. I see that one of forks of my library has an example for SD. I don't know whether it works fine or not, but I think it's at least some start point or a piece of code that you will be able to take inspiration from.
https://github.com/guigui9/ESP_VS1053_Library/blob/master/examples/SDMp3Player/SDMp3Player.ino

PS: Please do not forget to click the "star" for my project if you like it.

williamesp2015, 2018-07-30:
I appreciate  @baldram. Unfortunately the sound I converted is like a noise and prolonger than the original WAV file.

[~SOUND CONVERT.zip](https://github.com/baldram/ESP_VS1053_Library/files/2239931/SOUND.CONVERT.zip)

baldram, 2018-07-30:
@williamesp2015 I tried to use the file you provided me. 
The first thing I saw immediately was my IDE warning, which has informed me that a C/C++ limit has been exceeded.
[image]
But I tried to ignore it and build the project to see what will happen. It has ended up with an error as below. 
```
Linking .pioenvs/esp32dev/firmware.elf
/home/msx/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esp32dev/firmware.elf section `.dram0.data' will not fit in region `dram0_0_seg'
/home/msx/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 234944 bytes
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32dev/firmware.elf] Error 1
========================== [ERROR] Took 3.06 seconds ==========================
```
So I'm not sure how did you manage to compile that. But nevermind, let's go to the solution.
Probably if the file is too long in meaning of some IDE limit related to C/C++ it doesn't make sense to spend any additional effort on it. It is probably simply too long to avoid troubles. First of all the wav is uncompressed audio format. 
I would suggest to convert any mp3 file to c header, as mp3 is simply smaller. If you don't care about the stereo, I would also convert the file to mono, so again it will be smaller. In microcontrollers world memory is at a premium. So if you want to use some short sounds in your program (like button sound, notification or so), then it's still possible to decrease the size enough to make them part of the firmware. If it's something longer like whole mp3 song, then (the approach you already mentioned) it would need to be streamed from SD card, web or so.

But it's still easy to use the sound as C header file.
**Here is an example step by step**. It was **just made on the fly while answering your question**. 

I downloaded some random file from here: http://soundbible.com/2101-12-Ga-Winchester-Shotgun.html .

Then I decided to trim the file (ctrl+T) and remove silence (or almost silence) from the beginning and end. 
[image]

I manage to do it easily using free audio editor from https://www.ocenaudio.com/. I also converted it to mono (Edit > Convert Sample Type). As I observed one more potential issue 48kHz (instead of 44kHz which is equal to CD quality), I decided to reduce it (also via Convert Sample Type menu). But on the other hand from spec I see VS1053 supports up to 48kHz, so it should not be an issue. So maybe I didn't have to decrease this value.
[image]

Ok, further step. I exported the modified audio as mp3 with quality 112kbps. 
I would go even lower for such a gunshot sound (this is not an audiophile music). We need to think about the memory when programming microcontrollers.

So... finally executed `xxd -i gunshot2.mp3 gunshot2.h`.

What about the result? It works! :-)
Please find an attached zip with this mp3 and the output.
[gunshot2.zip](https://github.com/baldram/ESP_VS1053_Library/files/2242585/gunshot2.zip)

baldram, 2018-08-01:
@williamesp2015 I tested the `xxd` provided by you under Windows. It works correct.
The mp3 file from [gunshot2.zip](https://github.com/baldram/ESP_VS1053_Library/files/2242585/gunshot2.zip) has been converted successfully. There is a slide difference in file size (Windows: 177440 and Linux: 175072) because of the fact that text files created on DOS/Windows machines have different line endings than files created on Unix/Linux. But it doesn't matter.
The content inside is simply identical and it works. I uploaded the firmware to ESP32 board to check it.
So your `xxd` works fine!

The only thing is what I mentioned earlier, that it's better to use mp3 instead of wave, as the file should be small enough since microcontrollers have limited resources.

I will add a hint about the `xxd` to library documentation and then I will close this issue.
Is it ok for you?

baldram, 2018-08-03:
No answer. I consider the subject to be explained. Documentation has been extended. 
Closing the issue.

closed 2018-08-03 by baldram

tejarex, 2021-04-15:
i am not able convert .mp3 to .wav in c program can  one help me .

tejarex, 2021-04-15:
if there is any program is available please share me