baldram/ESP_VS1053_Library #23: Sdcard Not really an isssue

fredericplante, 2018-08-07:
I was wondering why there was no SDcard example?`

I was asking to know  you favorite FAT library or the one you usually use.

Great work by the way extracting that code from the ESP-radio example. :) I really appreciated.

baldram, 2018-08-08:
@fredericplante 
> I was wondering why there was no SDcard example?`

SD card is another peace of hardware. Of course VS1053 driver can be combined with that or anything else. Unfortunately I don't have any tested (myself) example. Of course such example could be placed in examples folder, but I would need to find time and test it well and then publish such example.
But the project is opened for pull requests. For instance the web radio example was implemented by someone else. If example is not provided by the community, maybe one day I will add SD card example myself as you are the second person asking for that. 
For now I can propose something. I see that my libary was forked to add exactly an example for SD card (I don't know why it was not proposed as a pull-request here, maybe I will ask for that). I also don't know whether it works fine or not, but I think it's at least some start point or a piece of code that you will be able to take inspiration from.
https://github.com/guigui9/ESP_VS1053_Library/blob/master/examples/SDMp3Player/SDMp3Player.ino

I hope it helps for now?

> I was asking to know your favorite FAT library or the one you usually use.

As I use PlatfomIO and this ESP_VS1053_Library is also designed to be used with PlatformIO, I will propose a library for this ecosystem. I would probably go for this one:
https://platformio.org/lib/show/322/SdFat

> Great work by the way extracting that code from the ESP-radio example. :) I really appreciated.

Thank you.

fredericplante, 2018-08-09:
Cool. So the SDcard part is done.

Next, I see that you have centralized all the pin management into one function. That is neat because it mean that I could use PCF8574 to handle the chip enabling and function pins. This might seam useless for an esp32, but I plan to use an esp8266 to do the media playing and on those device, every pins count.

Talk to you if I have forwards.

baldram, 2018-08-09:
> Next, I see that you have centralized all the pins management into one function.

Do you refer to VS1053 library, not to the SD card example? Can you be more precise and provide some more details?  
If by saying _"pins management into one function"_ you mean the constructor:
 `VS1053 player(VS1053_CS, VS1053_DCS, VS1053_DREQ)` ...
I'm not sure the library (the base code extracted from ESP-radio) supports PCF8574 expander. It will rather not work if about these pins. This is because these pins are about hardware SPI interface (that's why `SPI.h` library is required). 

Or you have some idea how to change the code in reasonable way with backward compatibility?

Unfortunately ESP8266 does not offer a sufficient number of pins. I understand what you mean by saying "every pins count". That's why I'm rather focused on ESP32, but as I have still a lot of ESP8266 devices I will use them for projects with very basic display, without display or so.

PS: The library is even older than Esp-radio. The VS1053 driver comes from here: https://github.com/maniacbug/VS1053/blob/master/VS1053.cpp . The radio project has provided a port for ESP devices.

fredericplante, 2018-08-10:
I have my idea, as soon that I'm done, I will give you news about it.

baldram, 2018-08-10:
> I have my idea, and all i'm saying is that the way you have built your library will 
> make it very easy to do pin handling from and I2C port. 
> As soon that I'm done, I will give you news about it.

I'm very happy to hear that and it's a pleasure for me that the library helps others. 
That's interesting what you do and I'm curious about the example with using PCF8574.

> Yet I have no idea if the process will be fast enough to drive the device correctly.

If it's successful I think we can point to your project from ESP_VS1053_Library documentation. 
I will create a section with the list of projects using the lib, like this one: https://github.com/vincegellar/Simple-Radio-Node .

baldram, 2018-08-12:
@fredericplante Shall we close the issue for now?

closed 2018-08-12 by fredericplante

baldram, 2018-08-13:
Thanks @fredericplante  :-)
When you have some interesting results please let me know and we can add a reference to your project in a list of projects using the ESP_VS1053_Library.
Thanks and good luck.