baldram/ESP_VS1053_Library #25: Improvement: make testComm method private

wmarkow, 2018-10-14:
I just thought that this method may be used very frequent to check if the communication with the VS105s chips is correct: in other words I just wanted to use this method to check if the VS1053 chip is connected.

I started to have a weired results (like silnce in my headphones ) when I used ```testComm``` in my project. It took me a while to find out what is going on: this method not only check if the VS1053 is connected but also tests the connection to the VS1053. It uses the **SCI_VOL** register to multiple save and read back a different values. It takes a bit time and also manipulates the volume.

I propose that this method is a private method and should be only executed from ```begin()``` method.

To make a simple test if the chip is connected I propose a solution introduced in #24.

wmarkow, 2018-10-14:
Ok - I know - the library is already released and changing the method modifier from **public** to **provate** is an API change, so the method should stay **public** in bugfix release.

baldram, 2018-10-16:
You're right. It should stay public. We can give the user to decide whether to use it or not. 
Then I think a good idea would be to add a comment to distinguish between `testComm()` and your proposed `isChipConnected()`. Both would be properly described and then there would be no doubts which one should be used according to current needs. 

> ...It took me a while to find out what is going on...

As you already analysed what this method does and how it works, maybe you would like to describe it and add a proper comment? Then the comment could also provide a suggestion that in some cases just calling the `isChipConnected()` is enough. What do you think?

Would you like to provide a pull request (with code comment only, then method would stay public)?

baldram, 2018-12-10:
@wmarkow We merged already some connected pull requests. What about this one? Will we keep it as it is and close the issue?

wmarkow, 2018-12-19:
To be honest I do not have much time right now to make that additional comment about ```testComm()``` (as you have suggested above).

We have agreed that the ```testComm()``` stay **public**.

I think we can close it.

closed 2018-12-19 by wmarkow

baldram, 2018-12-19:
Thanks! All your changes are already merged. I will release the library then soon with the latest changes.

Have you had chance to try the latest version?
It is possible (even not yet released) by using the dependency in `platformio.ini` directly from Github this way:
```
lib_deps =
    baldram/ESP_VS1053_Library
```