baldram/ESP_VS1053_Library #29: Cannot compile library

mrohner, 2018-11-11:
Tried to compile a program with this library and get many errors with IDE 1.8.3.
I used many of the ESP32 borads under Tools such as 'Wemos Lolin32' or 'ESP32 Dev Module'. I do not have a choice of the three microcontrollers you mentioned in the description.

In file included from C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.h:39:0,

                 from C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:33:

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'bool VS1053::testComm(const char*)':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:119:9: note: in expansion of macro 'LOG'

         LOG("VS1053 not properly installed!\n");

         ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:132:5: note: in expansion of macro 'LOG'

     LOG("%s", header);  // Show a header

     ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'void VS1053::begin()':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:156:5: note: in expansion of macro 'LOG'

     LOG("\n");

     ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'void VS1053::stopSong()':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:235:13: note: in expansion of macro 'LOG'

             LOG("Song stopped correctly after %d msec\n", i * 10);

             ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'void VS1053::softReset()':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:244:5: note: in expansion of macro 'LOG'

     LOG("Performing soft-reset\n");

     ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'void VS1053::printDetails(const char*)':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:254:5: note: in expansion of macro 'LOG'

     LOG("%s", header);

     ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp: In member function 'void VS1053::switchToMp3Mode()':

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\ConsoleLogger.h:22:60: error: 'ESP_LOGD' was not declared in this scope

         #define LOG(...) ESP_LOGD("ESP_VS1053", __VA_ARGS__)

                                                            ^

C:\Users\marku\Documents\Arduino\libraries\ESP_VS1053_Library-master\src\VS1053.cpp:278:5: note: in expansion of macro 'LOG'

     LOG("Switched to mp3 mode\n");

     ^

baldram, 2018-11-11:
Hi, thank you for feedback. 

It looks like it is not correctly used. 
As far I know someone has successfully run it with Arduino IDE, however this is a [PlatformIO](https://platformio.org/) library as described [here](https://github.com/baldram/ESP_VS1053_Library#why-platformio). 

Have you tried to compile it with PlafromIO? Here is a [library's page](https://platformio.org/lib/show/1744/ESP_VS1053_Library) in PlatformIO Libraries repository.
Here you will find [tutorials](https://docs.platformio.org/en/latest/tutorials/espressif32/arduino_debugging_unit_testing.html) on how to get started with PlatformIO.

The PlatformIO is a great tool which makes working with with Arduino, ESP and other micro-controllers much easier. It resolves dependencies for you automatically. I recommend to give a try.

Is this answer satisfying you? Can we close the issue?

mrohner, 2018-11-12:
Thanks for all the links. I give PlatformIO a try.

closed 2018-11-12 by mrohner