baldram/ESP_VS1053_Library #32: MIDI features / example

knopserl, 2018-12-28:
Thanks for this library, which I'm using with an ESP32. Seem to work fine, at least the MP3 demo with the peep tone.
I'm looking for the built in MIDI features and playing MIDI songs  (via the SD card) or programatically.
I have tried other VS1053B libs with MIDI examples without success.
I'm using the VS1053B MP3 Shield from geeetech and don't know how to send MIDI notes to whoch RX Pin or via SPI an dhow to activate the MIDI GM mode.
Maybe you can help me out?

THis is the module I'm currently using (it claims to have MIDI support)
https://www.aliexpress.com/item/MP3-Music-VS1053B-shield-board-with-TF-card-slot-work-with-UNO-MEGA/32649280181.html?spm=2114.search0104.3.15.27304502ECvItk&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10547_319_317_10548_10696_10084_453_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10059_10884_10887_100031_321_322_10103,searchweb201603_52,ppcSwitch_0&algo_expid=b00970c1-1881-4860-99cd-1a7402eab081-2&algo_pvid=b00970c1-1881-4860-99cd-1a7402eab081

baldram, 2019-06-03:
@knopserl I'm sorry I overlooked your question.
The library was not designed to support the MIDI, but theoretically it is possible as VS0153 should be able to handle this. I have never tried to do so. I rather focused on MP3, what you can see even in documentation. The line below ensures that the board switches from the (sometimes) default MIDI mode to the MP3 one.
`player.switchToMp3Mode();`

On the other hand something similar could be done to handle the MIDI, but you would have to look on some forums or in VS0153 specification, to find what to call to ensure that the board is in the MIDI mode. 

Sometimes, for some hardware, it is required to configure jumpers properly. Please see this for example: https://learn.adafruit.com/adafruit-vs1053-mp3-aac-ogg-midi-wav-play-and-record-codec-tutorial/midi-connections . If you add proper jumpers, the board will boot up in MIDI mode. 

I don't know what if about your case, but maybe something similar. 

You can also see an interesting comment in the beginning of this [code example](https://github.com/adafruit/Adafruit_VS1053_Library/blob/master/examples/feather_midi/feather_midi.ino). It says "Solder closed jumper on bottom!". So for sure there is something if about jumpers to handle the MIDI mode properly. 

But on the other hand maybe you already did some progress if about this topic? 
What is the current status?

knopserl, 2019-06-22:
@baldram long time ago, bcause I was busy with another ESP32 project. Now I'm back on the MIDI toolbox. I'm still looking for a good VS1053 lib which supports or provide MIDI functionality. MP3 examples are shown everywhere (seem to be the only well documented use case), MIDI is still a bit of a secret.
I'll not set the GPIO1 hard wirted to ground to enable real time MIDI but rather want to enable the VS1053 to play MIDI on demand (by command/SW). THere is a sequence which switches the chip to play MIDI (I'll do it via SPI), but as mentioned, I'm lookimg for a ESP32 proven lib and a good example.

baldram, 2019-06-24:
@knopserl Thanks for reply.

Please see what I found in the "original" version of this library (it is mentioned in the README).

Here is the source and you will find the RtMidi class what is missing in ESP_VS1053_Library.
https://github.com/maniacbug/VS1053/blob/master/VS1053.h

The MIDI part was not included in "Edzelf/Esp-radio" project when I was extracting the ESP compatible version of the library. Now it looks like the MIDI part could be extracted from the original maniacbug's project and added here (and ported to ESP if needed).
I have no time currently to test whether it works with the ESP. 
If you're able to do this, it would be great. Further you could create a Pull Request to add the missing code and some example. 

What do you think?

flossandmeditate, 2019-06-26:
@knopserl looks like we are hunting for the same thing, i started down the track with https://github.com/MajicDesigns/MD_MIDIFile but haven't had any success on the ESP32 - https://forum.arduino.cc/index.php?topic=620916.0 ... Might see about soldering some connections on the vs1053 to get MIDI mode forced....

knopserl, 2019-06-29:
@flossandmeditate lets see, I don't give up yet. The MIDI mode should be possible to be enabled via a SPI code sequence not requiring real time MIDI. Real time MIDI mode required GPIO0 and 1 to be in certaion states when powering up/reseting. I want to switch the VS1053b via SW which is SPI not requiring the mentioned pins soldered and connected.

baldram, 2019-06-30:
The missing piece could be [mentioned](https://github.com/baldram/ESP_VS1053_Library/issues/32#issuecomment-505159696) RtMidi class which can be ported to ESP (if needed) and added to the library. If about the hardware things (soldering or not) to enable the MIDI mode I keep fingers crosses.
I curious about the results so please let us know if you finally get it working.

knopserl, 2019-07-28:
@baldram RT MIDI by loding a usercode to the VS1053B works, it does then play MIDI notes.
These examples work as MIDI player through SPI or serial:
MP3_Shield_RealtimeMIDI.ino 
https://gist.github.com/microtherion/2636608
https://www.mikrocontroller.net/attachment/353944/MP3_Shield_RealtimeMIDI_demo.ino

baldram, 2019-07-30:
@knopserl That's good news. Thank you for testing this. 
Definitely ESP_VS1053_Library has to be extended with this. 
I have currently very limited time, but it will be handled soon I think.
If anyone in the meantime is able to deliver a PR with this extension, would be nice of course.
If not, that's also ok, I will include this stuff for sure ASAP.

baldram, 2021-06-18:
We get closer to resolve this one. An important milestone achieved thanks to this PR: https://github.com/baldram/ESP_VS1053_Library/pull/69. 

More details here: https://github.com/baldram/ESP_VS1053_Library/discussions/66.

Dr-Dawg, 2021-09-10:
While trying to figure out, why I get MIDI on the VS1053 using the library, but not on the VS1003, I stumbled across a bug in the loadUserCode function, which results in no patches being loaded at all. Will continue to discuss this in #66 Support loading patches.
For MIDI the effect is, that VS1053 has MIDI functionality in ROM and plays MIDI, while VS1003 is helpless without a patch.

Dr-Dawg, 2021-09-11:
@baldram: as you might have guessed by #82, I can provide a MIDI example pretty soon :-)

baldram, 2021-09-12:
That would be amazing @Dr-Dawg! 🤟

Dr-Dawg, 2021-09-16:
AFK dor 2 1/2 weeks. The MIDI example is ready, I will start a PR when I'm back, eventually including changes in the loadUserCode function, but I think it's fine with two parameters and the Bugfix-PR can be merged. I just checked the code of @maniacbug and he did the same, so it's "back to the roots" ;-)

baldram, 2021-09-16:
Thanks @Dr-Dawg for checking the solution in the "root" code! So let's merge it. 
When you're back, your MIDI MR is very welcome!

Dr-Dawg, 2021-09-17:
Ok, there's a little time left, so here we go..
Some notes about the MIDI feature and example: 
I just added one function, sendMidiMessage, to the library, the noteOn and noteOff functions are defined within the example. 
The reason for this is there are so many more MIDI commands beside noteOn and Off (e.g. Aftertouch, Pitch bend, Modulation Wheel, Expression,..) that this would IMHO exeed the scope of the library.
The vast amount of possible MIDI commands deserve their own library/header file. Maybe it already exists and can be adopted to this library?

There seems to be an issue with VS1003 devices not reacting / running to an infinite loop after loading the MIDI patch, see http://www.vsdsp-forum.com/phpbb/viewtopic.php?t=1151
The reason for this is the DREQ pin is not raised by the device for whatever reason. The solution would be to comment await_data_request();  in the sendMidiMessage function. 
It also happened to me, unfortunately not reproducable. I kept await_data_request in for the sake of preventing an input buffer overrun, even if it is unlikely to happen in RT-MIDI.
Btw cheers to user kivig in the above mentioned VLSI page. After small modifications his code was actually the first one I tested to produce a sound on my VS1003 device!

Have fun! I'm curious to hear whether anyone tested the example :-)

baldram, 2021-09-17:
Thank you! I will look into more carefully into this after the office hours. 

>  that this would IMHO exceed the scope of the library.

I agree, and I have the same feeling. I would even think whether it is possible to extract the MIDI related stuff to separate `ccp`/`h` files as an optional library module. But, on the other hand, it's one of the capabilities of the VS1053 chip. So it's nice to have (but probably with limited scope is OK, as the library itself acts as a driver, enables the essential feature, and lets the user go further according to own requirements - so those are in the "example" implementation). Looks good. 

> Have fun! I'm curious to hear whether anyone tested the example :-)

😏

stanleyseow, 2022-05-22:
This article ( i tested successfully running a VS1003 ) using RTMIDI ) on generic VS10xx boards ...

https://diyelectromusic.wordpress.com/2021/01/31/arduino-midi-vs1003-or-vs1053-synth/

baldram, 2022-05-22:
@stanleyseow Actually all work is done here in this MR" https://github.com/baldram/ESP_VS1053_Library/pull/83
We're only missing someone who will test it and run it using his breadboard and setup. 

PS: I need to re-setup my workshop and then test this stuff. Sorry @Dr-Dawg  it takes so long.