baldram/ESP_VS1053_Library #34: Issue depending of the "esp 8266 Community" version for Aduino IDE

philippedc, 2019-05-31:
Hi I've written a program that use your library with many enthusiasms. So much that I've been requested to build several web radios. BUT now I have the program aborting at the line : 
player.playChunk(mp3buff, bytesread);
When I comment this line everything is ok. However once uncommented the esp reboot infinitively.
I've spend half a day to understand the reason why, with no success....
Except that between time I built the first web radio and now, the "esp 8266 Community" (It is the package we must install when we want to use Wemos D1 mini ( or others) on an Arduino IDE.) version were 2.4.2. Now it is 2.5.x and none of 2.5.x version is working.
I'm sorry, I'm not strong enough to understand why.

baldram, 2019-05-31:
Hi Thanks for using the library.
To be honest I have no experience with Arduino IDE. As you can read from the [README](https://github.com/baldram/ESP_VS1053_Library/blob/master/README.md), this library was designed be used with the PlatformIO (https://platformio.org/) which helps to develop application with pleasure. It helps to resolve different 3rd party library dependencies automatically etc. What is the most important, it gives you an integration with your favourite IDE, like: CLion, Microsoft's Visual Studio Code, Atom, Eclipse IDE for C/C++, etc. 
If about the library itself, as you see in [Credits](https://github.com/baldram/ESP_VS1053_Library/blob/master/README.md#credits), it was extracted from the [Esp-radio](https://github.com/Edzelf/Esp-radio) project. If about the other one mentioned there, I see that the author of Esp-radio got inspiration from other project developed earlier.

Why I point to Esp-radio. I extracted the "core" driver code and created a library based on this. 
But mentioned Esp-radio can be considered as a practical example. Maybe you will help you to figure out what a problem is. 

If I properly understand your problem description, the problem is that there is some conflict with the other 3rd party library you had to install to use Wemos D1. I'm not sure what the reason for this, as Wemos is just another ESP8266 and rather doesn't require any specific libraries. But if so, I haven't met such problem and I don't know that package and especially in context of Arduino IDE.

If you say about problem with `player.playChunk(mp3buff, bytesread)`, it's the most important moment of the program when you want to get the sound :) So for sure it's the most critical moment.
But I have no idea what's wrong when you say just "the program aborting at the line".

I can't promise I will have any clue, but if I could see the log, stacktrace, anything...

PS: Is your project published on Github or described somewhere on the Internet?

philippedc, 2019-06-01:
hi, many thanks for your answer. I did not know about platformio until yesterday :)
Nevertheless I like to have a look to libraries I expect to use, for instance if it is simple and understandable. Most of it is register writes with SPI, so it should not have matters of transportability. In fact the program compiles with no errors. It is "just" rebooting once the instruction player.playChunk(mp3buff, bytesread) is reached.
I have not published it as I'm not the full creator of the program, it is only an extension of your library example. I would like to add the functionality to record an audio message with a smartphone that is read on the web radio. 
Attached the .ino program, you have to replace the .gif extension. Sorry most of comments are in French....
[image]

philippedc, 2019-06-01:
... I've came back to the "esp 8266 Community" package 2.4.2, it works fine again. I'm still testing different ways to solve a kind of instability during the power switch on. Earlier I could have the VS1053 not ready and/or the ESP. That required to switch off and on several times to get the radio playing. With this version a hard reset of the VS1053 is performed in the setup, that partially solves the problem, sometime the ESP may be lost during the wifi connect.

baldram, 2019-06-02:
When you say "esp 8266 Community" package, do you mean something different from this: https://github.com/esp8266/Arduino ? 
What is the GitHub link to the package you mean?

When it is rebooting, do you get any debugger information? 
Once you set logging level to "debug", what does the console window show once it's rebooting?

wmarkow, 2019-06-02:
>Except that between time I built the first web radio and now, the "esp 8266 Community" (It is the package we must install when we want to use Wemos D1 mini ( or others) on an Arduino IDE.) version were 2.4.2. Now it is 2.5.x and none of 2.5.x version is working.

@philippedc, please confirm:
1. with "esp 8266 Community" 2.4.2 everything was working correctly.
2. it stoped  working when you use "esp 8266 Community" 2.5.x

philippedc, 2019-06-03:
Hi, yes I confirm 

-------- Message d'origine --------

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/baldram/ESP_VS1053_Library/issues/34?email_source=notifications\u0026email_token=AMDSNSV3GG43NC3OGWZWIJDPYQZ7XA5CNFSM4HR3SZQKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODWX6FUA#issuecomment-498066128",
"url": "https://github.com/baldram/ESP_VS1053_Library/issues/34?email_source=notifications\u0026email_token=AMDSNSV3GG43NC3OGWZWIJDPYQZ7XA5CNFSM4HR3SZQKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODWX6FUA#issuecomment-498066128",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

baldram, 2019-06-03:
@philippedc I'm not sure you noticed. Two different people asked you something in this thread. 
You probably overlooked my questions and haven't answered any of them.
Could you please provide an answer? 
Here are questions again: https://github.com/baldram/ESP_VS1053_Library/issues/34#issuecomment-498057682

philippedc, 2019-06-03:
Excuse me, yes I confirm for @wmarkow.

I will confirm for you as soon as I get a pc.

Thanks

philippedc, 2019-06-03:
@baldram : to be able to compile on WEMOS D1 Mini, I applied the procedure described many times on internet, for instance this one : https://randomnerdtutorials.com/how-to-install-esp8266-board-arduino-ide/
I will reinstall the "esp 8266 Community" latest version package to copy the console message,

baldram, 2019-06-03:
@philippedc Thanks. Then I see, you use exactly https://github.com/esp8266/Arduino what is fine.
Now I keep fingers crossed and hopefully the problem will be gone once you reinstall the ESP8266 Arduino IDE. If not, then the console log maybe will show some details. 

(A hint for [enabling debug](https://github.com/baldram/ESP_VS1053_Library#logging--debugging): `build_flags = -D DEBUG_ESP_PORT=Serial`. 
Here [some link to docs](https://github.com/esp8266/Arduino/blob/2eb5b5600a3631cdfb9ddd1150505d82e38f3406/doc/Troubleshooting/debugging.rst#debug-port) what seems to be relevant if about the Arduino IDE).

PS: Maybe in the meantime @wmarkow have some ideas...

philippedc, 2019-06-03:
@baldram Here what I have done, on a new wemos D1 mini not wired : 
1- install package 2.4.2 : compile OK, run OK
2- install package 2.5.2 as an update : compile OK, do not run. attached 2 screen captures : IDE after compilation, and console during running.
3- install package 2.4.2 over the 2.5.2 : compile OK, run OK
[image]
[image]

baldram, 2019-06-03:
> Here what I have done, on a new wemos D1 mini not wired

@philippedc If it's not wired, it won't work and it's correct behavior. It looks for VS0153 and if not found, it should fail. Sometimes similar  exception like yours I had when the wiring was incorrect, like one cable not connected properly or so. 

The debug you showed me is from your program. I don't see any log information from the library. 
Have you defined `DEBUG_ESP_PORT` constant to Serial ? Please see [this file](https://github.com/baldram/ESP_VS1053_Library/blob/master/src/ConsoleLogger.h), as specified, it expects this value to be set. 
 Then it should show some information. As you [see here](https://github.com/baldram/ESP_VS1053_Library/blob/master/src/VS1053.cpp), a lot of things is being logged. When you look at the source code, you can search for the phrase "LOG(".
If the logging is enabled, I would expect to see some log information as defined in VS1053.cpp.
What you showed me is only about your part, the radio stuff, so it says nothing to me.

Would you please try to get and show some logging information as mentioned above?

wmarkow, 2019-06-03:
@philippedc, from your stacktrace from ESP8266: there is something like **epc1=0x40207a7e**. This is the address of instruction where the programm stuck. Please use **objdump** utility (it is a binary programm that comes together with your toolchain) to find out what is the source code line related to this address. From the command line you need to call something like:

[path to the ESP8266 compiler]/bin/objdump.exe -e [path to your ELF file] 0x40207a7e

**path to the ESP8266 compiler** it is the path where your ESP8266 compiler is installed. I'm not so familiar wth Arduino IDE (I'm using Eclipse and Arduino [Sloeber](https://eclipse.baeyens.it/) plugin) but you need to find, where this compiler is installed.
**path to your ELF file** is where the binaries of your project ar built.
You can google about how to use *objdump*. Of course the ELF file must match the the HEX file that is loaded to your ESP8266 device.
Please use **objdump** and post the result here, meybe we will find something interesting.

Moreover, I have tried my [webradio](https://github.com/wmarkow/arduino-sandbox/tree/webradio) project and it seems to work fine (I have noticed that sometimes I need to reattach my VS1053 device by pulling of the power supply) with **esp8266-2.5.0** and the latest **ESP_VS1053_Library** library - but I'm using **NodeMCU** board. I have never observed the crash.

Last comment of @baldram is also interesting. Please pay attention also to this.

baldram, 2019-06-09:
How are you doing @philippedc? Any progress?

philippedc, 2019-06-10:
Hi @baldram I'm traveling right now, I will be installing a Piggott wind turbine in Gambia, I will be able to make now tests with your recommandations in July.
thank you for patience

baldram, 2019-06-10:
Hi @philippedc , thanks for feedback. Good luck then and waiting for your response when you're back.

baldram, 2019-07-13:
Hi @philippedc how about progress? Just asking ;-)

philippedc, 2019-07-15:
Ho @baldram, here it is, but even so it is not clear to see anything.....
[image]
[image]
[image]

philippedc, 2019-07-15:
I've found an avr-addr2line.exe but no idea for the moment where is the "path to your ELF file"
[image]

wmarkow, 2019-07-16:
@philippedc you have found a wrong **avr-addr2line.exe**. The you you found seems to be for a standard Arduino Uno or something. You need to find a toolchain for ESP8266 - this has a different names, something like **xtensa-lx106-elf-addr2line.exe**.

On my Windows machine it is located here:
```
"C:\\Users\\<user name here>\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc"
```
so it may be similar for you.

In Arduino IDE you can also enable verbose messages output during sketch compilation. You will see then a lot of useful information during compilation - also the path to compiler. You can enable this option by checking **Show verbose output during compilation** option enabled in **File -> Preferences**.
The output log will also show you where the ELF file is stored. In my example it is here:
```
"C:\\Users\\<user name here>\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-size" -A "C:\\cygwin64\\tmp\\arduino_build_460504/WebRadioDemo.ino.elf"
```
@philippedc read again [my comment from 3 Jun](https://github.com/baldram/ESP_VS1053_Library/issues/34#issuecomment-498435491) to learn how to use **add2line**. In your last example you need to query for **0x4020a176** address. The **addr2line** should output the source file name and the line number of the last executed line (before crash).

philippedc, 2019-07-20:
EUREKA !!!! 
Many thanks for your patience. I assume I still have to learn a lot before I become an expert....
Here is the answer : <users home>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI/SPI.cpp:430

which is the *fifoPtr of the followed while instruction :

    uint32_t * fifoPtr = (uint32_t*)&SPI1W0;
    const uint32_t * dataPtr = (uint32_t*) data;
    uint32_t dataSize = ((size + 3) / 4);

 while(dataSize--) {
        *fifoPtr = *dataPtr;
        dataPtr++;
        fifoPtr++;
    }
I compared with the SPI.cpp file in version 2.4.2, this part is exactly the same. The problem is before of course.

baldram, 2019-07-20:
Thanks @philippedc for sharing the results. Thanks to @wmarkow for his assistance and very professional advices. Such threads hidden inside Github issues log are mine of information. I often browse closed issues to find interesting things and learn something. Hopefully this one will help someone too in the future.

The VS1053 library seems to work itself as confirmed by @wmarkow in thread [this comment](#issuecomment-498435491), the problem is with SPI.cpp or something else which is connected to it.
I wish you good luck ;-) to find the root cause. 

Do you think @philippedc we can close the issue at this point?
Of course even the issue will be closed, it's still possible to add comments and share the information about the final solution (to help someone with similar problem in the future).

So do you think we can close the issue at this point?

philippedc, 2019-07-20:
@baldram yes you can. Many thanks for your contribution and your library. Thanks to @wmarkow for his pedagogic way to explain what to do.

My web radio is working fine with 2.4.2 esp8266 IDE package, so it has not been the first time a bypass has to be found to prevent against a bug.

wmarkow, 2019-07-20:
I'm happy that I could help. However I have took a look at the **SPI.cpp:430** in esp8266 **2.4.2** and **2.5.2** and it is as @philippedc has mentioned: the code around this line has not changed.

@philippedc, please do a test for me. Change your code from
```
// si tout va bien :)
  if( client.available() > 0 ) {
    uint8_t bytesread = client.read(mp3buff, 32);
    player.playChunk(mp3buff, bytesread);
  }
```
into
```
// si tout va bien :)
  if( client.available() > 0 ) {
    uint8_t bytesread = client.read(mp3buff, 32);
    if(bytesread <=32) {
        player.playChunk(mp3buff, bytesread);
    } else {
     Serial.println("Oops, something wrong here!");   
    }
  }
```
so the ```bytesread``` variable will be checked if it contains a value lower than or equal to **32**. I'm not quite sure if something bad happens when a value of this variable is much higher than 32 or 64 (like 255 is the highest value possible in your code).
Please test this code with ESP8266 version **2.5.2**. It would be nice if you post here the result.

wmarkow, 2019-07-20:
@philippedc , you can also try something what I have in my webradio application (which seems to be working correctly with esp8266 **2.5.0**):
```
if (client.available() >= 32)
{
    uint8_t bytesread = client.read(buffer, 32);
    player.playChunk(buffer, bytesread);
}
```
I'm always sending 32 bytes of data.

philippedc, 2019-07-21:
@wmarkow I've tested both. The "Ooops" message is never displayed, either the exception is done before, or everything is right for the test.

Note that esp8266 package 2.5.0 is no more available in platforms menu. There 2.4.2 and before, or 2.5.2
Otherwise I would have tested this version if I get it working as for you.

wmarkow, 2019-07-21:
Thanks @philippedc , however I'm running out of ideas :(

In Arduino IDE, when you open your sketch you can choose *Tools* (on the application menu at the top) and try to change the setting for **lwIP Variant** to something else. This **lwIP** is some kind of additional library which is used by ESP8266 and it is added at the compilation time. I see that there are eight different implementations possible to choose. I have also checked that especially **lwIP v2** has changed a lot between esp8266 **2.4.2** and **2.5.2** (however v1.4 has not changed so much).
**v1.4 Compile From Source** may not work because you need to provide sources for this library manually for your sketch.

philippedc, 2019-07-24:
@wmarkow hi, I have tested with several IwIP options, but each time I get the same issue.

wmarkow, 2019-07-25:
Ok, so now I have no idea what could break it. We know for sure that it works with esp8266 **2.4.2** and it stopps working with **2.5.0** (as you wrote in your forst comment that it doesn't work with 2.5.x). So we can assume that there is some change between **2.4.2** and **2.5.0** which makes that it doesn't work for you. The last idea is to find that change - actually need to find a specific commit. It can be easy achieved by:
 1. we know that **2.4.2** works
 2. we know that **2.5.0** doesn't work
 3. you need to clone the sources from https://github.com/esp8266/Arduino
 4. pick a commit which is directly in the middle between **2.4.2** and **2.5.0**. Lets call this commit **Commit X**.
 5. copy the relevant sources tree into your arduino **AppData** folder. Mine is as 
```C:\Users\wmarkowski\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\```
 It doesn't metter now that the folder is named 2.5.2. Let's Arduino IDE to think that it has the 2.5.2 version installed
 6. build your project with Arduino IDE
 7. upload the code to the board
 8. make the test if that works. Now it depends on the result:
    1. if it works then it means that it is broken between **Commit X** and **2.5.2**. Choose a new commit in the middle between **Commit X** and **2.5.2** and go to step 5. Proceed to narrow down to finally find the commit that breaks the functionality.
    1. if it doesn't work then it means that it is broken between **2.4.2** and **Commit X**. Choose a new commit in the middle between **2.4.2** and **Commit X** and go to step 5. Proceed to narrow down to finally find the commit that breaks the functionality.

In around 10 iterations you can find a commit of interests. You do not have to do this all manually. Git supports this mechanism to make it easier for you. It is called **git bisect**.

As a good start take a look here (of course you can google more):
https://git-scm.com/docs/git-bisect
https://flaviocopes.com/git-bisect/
https://stackoverflow.com/questions/4713088/how-to-use-git-bisect

baldram, 2021-06-12:
This seems to be no longer valid issue. People using Arduino IDE are successfully using a library what I see from recent PRs or external projects dependent on ESP_VS1053_Library.

closed 2021-06-12 by baldram