baldram/ESP_VS1053_Library #37: 'VS1053 not connected properly'

skoyaDEV, 2019-07-29:
Hi, I am working with ESP32 Devkit and VS1053. I've downloaded these libraries and everything is working fine ( I have the sound on the output ) but when I try to launch isChipConnected() it tells me that it is not connected properly. What might be the problem?

baldram, 2019-07-30:
The `isChipConnected()` method was introduced by @wmarkow as a result of this discussion #24 .
Please find some more details describing the idea of this method.

You ask what might be the problem... 
It could be caused by some hardware failure like: the chip is not correctly powered up, not correctly connected and so on. Sometimes it's wiring issue, but you mentioned that "everything is working fine" and you have the sound on the output. I had sometimes the issue that cables were disconnecting occasionally (the plug did not work well, or the socket was too loose). 
It's hard to say, as different issues may happen...

wmarkow, 2019-07-30:
It may indicate a hardware issue: power supply, cables not soldered correctly (disconnecting occasionally), to long wires and so on.

```isChipConnected()``` reads the status register from the VS1053. If the value is **0x00** or **0xFFFF** then you can have something wrong with MISO line of the SPI interface. When the MISO line is not connected at all (so it "hangs in the air") then - depending on the specific microcontroller (or if the MISO is pulled up or not) - the read bits by SPI is usually zeros (which gives a value of 0x00) ore ones (which gives a value of 0xFFFF).
In the scope of status register - accroding to the datasheet of VS1053 - both values 0x00 and 0xFFFF are not possible to have.

baldram, 2019-09-02:
@skoyaDEV I close the issue as a lot of suggestions were provided.
But please comment and tell whether it helped you and which exactly suggestion was helpful?

closed 2019-09-02 by baldram