baldram/ESP_VS1053_Library #38: no sound, even after player.testComm("testing...\n ") and player.isChipConnected() have passed

GalBrandwine, 2019-08-04:
hello, im trying to use the  [ESP_web](http://educ8s.tv/esp32-internet-radio) sketch, and all the tests have passed.

the esp32 is connected to WiFi, the station it tryes to read is valid (tryied myself via chrome),but still no sound coming out of the vs1053 board..

any suggestion?

thanks, Gal

baldram, 2019-08-04:
Hi, thanks for using the library.
I would suggest connecting your board in the way similar to my tests here: https://github.com/baldram/ESP_VS1053_Library/issues/1#issuecomment-313163772 . 
Then please adjust the example from here: https://github.com/baldram/ESP_VS1053_Library/blob/master/examples/Mp3PlayerDemo/Mp3PlayerDemo.ino
By adjust, I mean to set up the example in the same way as you wired your board, for instance:
```
#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4
```
Please double check if all cables are connected properly and so on.
If it works, further you can try something similar with your sketch.

Good luck. Please let me know about the result.

baldram, 2019-09-02:
As mentioned earlier, a lot of wiring examples with photos were provided by me in [this thread](https://github.com/baldram/ESP_VS1053_Library/issues/1#issuecomment-313907348).
If everything is properly connected it should play smoothly (even one cable wrong causes no-sound).

I close this issue for now, but please comment and share your final solution by telling what was wrong with wiring on your side...

closed 2019-09-02 by baldram

GalBrandwine, 2019-10-14:
Hey, yeah it was a wiring problem eventually..

Thanks for the help!