baldram/ESP_VS1053_Library #39: Play file from SPIFFS

Voha888, 2019-08-21:
I uploaded mp3 files to spiffs. When I try to reproduce them, I get an error. Here is the code:
```
uint8_t mp3buff [34];
File file_SPIFFS;
file_SPIFFS = SPIFFS.open ("/ 02.mp3", "r");
file_SPIFFS.readBytes ((char *) mp3buff, 32);
player.playChunk (mp3buff, 32);
```
Error related to data size:
```
3ffeeaf8 40208e89
3fff0080: 3ffe85d3 3ffeea6d 3ffeeaf8 3ffeec18
3fff0090: 00000040 3ffeea6d 3ffeeaf8 402087fc
3fff00a0: 00000000 3ffeea6d 3ffeeaf8 402010ef
3fff00b0: 3fffdad0 3ffee8cc 3ffeeaf8 40201f0d
3fff00c0: feefeffe 00000000 3ffeebe8 4020a654
3fff00d0: feefeffe feefeffe 3ffe8508 40 100 581
<<< stack <<<

  ets Jan 8 2013, rst cause: 2, boot mode: (3.6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ ld
```

baldram, 2019-08-21:
Hi, 
unfortunately you provided not enough information to say anything.

First of all the stacktrace you provided doesn't show much. 
Would you please find the line where the problem is using addr2line utility?
Some more information would be needed. 
Would you please answer questions below?

1) Please find out what is the source code line related to this address where the issue is (using addr2line utility).
Here is a thread which describes with more details what to do. Please follow this and provide a result:
https://github.com/baldram/ESP_VS1053_Library/issues/34#issuecomment-498435491
Here is additional explanation: https://github.com/baldram/ESP_VS1053_Library/issues/34#issuecomment-511705008 

2) What kind of hardware do you use?

3) What about wiring details? I mean something like i described here: https://github.com/baldram/ESP_VS1053_Library/issues/1#issuecomment-313163772

4) Do you use PlatformIO to compile and run your program? This library is designed to be used with PlaformIO ecosystem. 

5) What about 3rd party libraries you use? What are the versions and so on? What about ESP arduino core version you use?

baldram, 2019-08-25:
Hi @Voha888 . Any results? Previously described instruction points to the place where the problem is. In case it happens inside the VS1053 library, additionally you could enable DEBUG mode to collect some helpful information. The easiest way is with PlatformIO, it is described here: https://github.com/baldram/ESP_VS1053_Library#logging--debugging .
You can also enable `DEBUG_ESP_PORT=Serial` in the other way, some [more info here](https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html#debug-port).

So first of all please use `addr2line` utility to get human friendly details for your stacktrace, as mentioned in [previous answer](https://github.com/baldram/ESP_VS1053_Library/issues/39#issuecomment-523502151) and provide details for the other questions.

baldram, 2019-09-02:
The error is thrown probably in the user's code. Without additional information asked in previous comments it's hard to investigate what is a real root cause. For now, I close this issue.

closed 2019-09-02 by baldram

Voha888, 2019-09-02:
When I create the array in format:
```
unsigned char  mp3buff[]={
  0x49, 0x44, 0x33, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x23, 0x54, 0x53,
  0x53, 0x45, 0x00, 0x00, 0x00, 0x0f, 0x00, 0x00.....
```
Then change it for my purposes - all works, but in other ways:
`unsigned char  mp3buff[32]={0};`
or
`unsigned char  mp3buff[32];`
I have the `Exception(9)`

_Sorry for not responding, I tried to find an answer or other solution_

baldram, 2019-09-03:
Thanks for response.

Yeah, there was a [thread about converting data](https://github.com/baldram/ESP_VS1053_Library/issues/18) to unsigned char.

I'm not sure you've seen examples [here](https://github.com/baldram/ESP_VS1053_Library/tree/master/examples/Mp3PlayerDemo), where SampleMp3.h has `unsigned char sampleMp3[]` .

There is also a very simple [example of web radio](https://github.com/baldram/ESP_VS1053_Library/tree/master/examples/WebRadioDemo), where the stream is directly taken from the Internet and the array is like this: `uint8_t mp3buff[32]`. It works. The radio is super simple, just add ring/circular buffer and it should play smoothly. 

I hope it may be helpful...

Voha888, 2019-09-03:
Sorry for my bad English, apparently you did not understand me. What I wanted now works for me. But the array has to be manually filled with zeros, I don’t know what is connected with something. In theory, auto-completion of the array should work when specifying` {0}`, or simply there should be no errors when specifying the size of the array. One specialist told me that this could be due to the fact that the array is located on the stack, and I'm trying to use it in a function, but I'm still new to programming, so I can not understand the reason for this behavior of the program

playground, 2021-03-07:
Hi @baldram @Voha888, I'm trying to do a similar thing too.  I want to create a mp3 file on ESP32 based on the response I get from https://texttospeech.googleapis.com/v1/text:synthesize 
`{
    "input": {
        "text": "Hello world!"
    },
    "voice": {
        "languageCode":"en-gb",
        "name":"en-GB-Standard-A",
        "ssmlGender":"FEMALE"
    },
    "audioConfig": {
        "audioEncoding":"MP3"
    }
}`

which returns the encoded audio for "Hello world!"

I decoded the response from text:synthesize api call
 
    obj = JSON.parse(response);
    mp3Data = obj["audioContent"]; 
    unsigned char b64_decipher[32] = {0}; 
    mbedtls_base64_decode(b64_decipher,sizeof(b64_decipher),&out_len,(unsigned char*)mp3Data.c_str(), mp3Data.length());
    file = SPIFFS.open("/helloworld.mp3, "w");
    file.write((uint8_t *)b64_decipher, out_len);
    file.close();

    mp3file = new AudioFileSourceSPIFFS("/helloworlld.mp3");
    id3 = new AudioFileSourceID3(mp3file);
    id3->RegisterMetadataCB(MDCallback, (void *)"ID3TAG");
    out = new AudioOutputI2SNoDAC();
    mp3 = new AudioGeneratorMP3();
    mp3->begin(id3, out);

But I'm getting 'Decoding error 'lost synchronization' https://github.com/earlephilhower/ESP8266Audio/issues/143#issuecomment-792110015

I think I'm not creating the mp3 file properly.  Any idea what I'm doing wrong?

baldram, 2021-03-08:
Hi, unfortunately, no idea. What I would suggest is to take a static mp3 from examples and test whether your code works anyway. So in line where you take data from API, use that static file with "demo" sound. If it works, it's a matter of comparing and analysis what kind of data that API provides. This might be also an issue with SPIFFS and the way of data handling what is not in the scope of this library, which is a sound device driver.