baldram/ESP_VS1053_Library #41: Decoding time

fabitom, 2019-11-28:
Please add support for check and reset decoding time by SCI_DECODE_TIME. Thx

baldram, 2019-11-29:
@fabitom The ESP_VS1053_Library is a core driver code extracted from [this project](https://github.com/Edzelf/Esp-radio). 
Do you see there an implementation for mentioned SCI_DECODE_TIME support? Are you able to point me to the desired part of code? 
I've also checked [maniacbug/VS1053](https://github.com/maniacbug/VS1053/search?q=SCI_DECODE_TIME&unscoped_q=SCI_DECODE_TIME) which was used as a base to port the driver for ESP devices. I see there only a constant like this, but I'm not sure this is what you are asking for?

fabitom, 2019-11-29:
It's simply read/write register, write_register(0x4,0)/read_register(0x4)
According to VS1053b datasheet user need to decide when the track starts. 
Here is in currentPosition():
[https://github.com/mpflaga/Arduino_Library-vs1053_for_SdFat/blob/master/src/vs1053_SdFat.cpp](url)

baldram, 2019-11-29:
The link above didn't work, but I looked into the project you pointed to. I've also had a look earlier and found [this one](https://github.com/HuangDave/MP3/blob/1f271460f0bc647875617149967f8418eb3c8121/L5_Application/MP3/Drivers/VS1053B.cpp#L144), where a simple method is provided.

So the goal would be extended the API with two methods.

Now I try to understand the problem to design the API with your help and find the proper name for new methods.
The first one could be easy, something like this:
```
// return the milliseconds offset of stream played
uint32_t getCurrentPosition();
```
Does it look ok to you?

If about the one to reset decoding time... it could be something like this:

```
// Reset the decode and bitrate from previous playback (SCI_DECODE_TIME)
void clearDecodeTime();
```
This name is a bit enigmatic, but I might be wrong. Optionally it could be called like: 
`void resetDecodeAndBitrate();`
...following the comment I found in the project you refer to: "`// Reset the Decode and bitrate from previous play back`".

So now the question is whether adding two methods is what you expected?
Which name would be more suitable in your opinion?

PS: I also had a look to [documentation here](https://mpflaga.github.io/Arduino_Library-vs1053_for_SdFat/vs1053___sd_fat_8h.html#ae70a5ef67f28bbbe578eca237450e3c1). 
It is said _"In that case the new value should be written twice to make absolutely certain that the change is not overwritten by the firmware."_ I'm going to do it. I see the same if about `HuangDave/MP3`, even `mpflaga` (the project you pointed to) has it only documented but if about implementation I'm not so sure...

fabitom, 2019-11-29:
Oh, sorry for broken link.
According to VS1053b datasheet: "current decoded time is shown in this register in full seconds", so IMHO you should return in seconds, (yes I know than "Arduino_Library-vs1053_for_SdFat" converts the value to milliseconds), and the name IMHO should point to time, maybe just simple DecodeTime/getDecodeTime and clearDecodeTime ?
And yes, the new value should be written twice, but written once works in my case ;)

baldram, 2019-11-29:
Ok, I've provided the change, but it's not yet merged to the master branch. 
It requires testing now. I will try to find time for this on Sunday. I will also need to find a box with ESP boards ;)
When successfully tested and merged, I will release a new version which will further show up in [PlatformIO Libraries Registry](https://platformio.org/lib).  

In the meantime would you please have a look at the change whether it's fine?  
The pull request is here: https://github.com/baldram/ESP_VS1053_Library/pull/42/files .

fabitom, 2019-11-30:
Thanks, you have forgotten return in getDecodeTime :)

baldram, 2019-11-30:
Ha, ha, good catch! :) I too much focused on comment and method name ;-)
As mentioned I will try to find time tomorrow to test it with ESP board when find one.

PS: Btw. I don't know yet how to implement for CPP Arduino something like JUnit to avoid such stupid mistakes. In Arduino world unit tests are not super popular.

fabitom, 2019-11-30:
PS. Maybe you will find some more time 👍 to get track info from SCI_AUDATA, below I is my code to get current track information:
```ruby
const char *afName[] = {
  "unknown",
  "RIFF",
  "AAC ADTS",
  "AAC ADIF",
  "WMA",
  "Ogg",
  "FLAC",
  "AAC MP4",
  "MIDI",
  "MP3",
  "MP2",
  "MP1",
};
int GetTrackFormat(uint16_t frm) {
	if (frm == 0x7665) {
		return 1;
	} else if (frm == 0x4154) {
		return 2;
	} else if (frm == 0x4144) {
		return 3;
	} else if (frm == 0x574d) {
		return 4;
	} else if (frm == 0x4f67) {
		return 5;
	} else if (frm == 0x664c) {
		return 6;
	} else if (frm == 0x4d34) {
		return 7;
	} else if (frm == 0x4d54) {
		return 8;
	} else if ((frm & 0xffe6) == 0xffe2) {
		return 9;
	} else if ((frm & 0xffe6) == 0xffe4) {
		return 10;
	} else if ((frm & 0xffe6) == 0xffe6) {
		return 11;
	} else {
		return 0;
	}
}
uint16_t ReadVS10xxMem(uint16_t addr) {
	playerV.write_register(0x7, addr); //SCI_WRAMADDR
	uint16_t reg = playerV.read_register(0x6); //SCI_WRAM
  return reg;
}
void ShowLog(void) {
  if (((playCounter % 600) == 0) && (playerV.isChipConnected())) {
    uint16_t seconds = playerV.getDecodedTime();  
    uint16_t track_info = playerV.read_register(0x5);  //SCI_AUDATA
    int rate = (track_info >> 1) * 2;
    int stereo = ((track_info << 15) >> 15);
    uint16_t h1 = playerV.read_register(0x9); //SCI_HDAT1
    uint32_t byteRate = ReadVS10xxMem(0x1e05); //PAR_BYTERATE if flac byteRate *= 4
    const char* TrackFormat = afName[GetTrackFormat(h1)];
    if (strcmp(TrackFormat,"Flac") == 0) {
      byteRate *= 4;
    }
    Serial.printf("\rTrack time: " "%02d:%02d:%02d" " Heap: %d/%d" " Rate: %dHz" " Ch: %s" " kb/s: %1.1f" " %s ", 
    seconds / 3600, 
    seconds / 60, 
    seconds % 60, 
    ESP.getHeapSize(), 
    ESP.getFreeHeap(), 
    rate, 
    (stereo == 0) ? "Mono" : "Stereo", 
    byteRate * (8.0/1000.0),
    TrackFormat);
  } 
}
```

baldram, 2019-11-30:
Ok, so if I see correct, `playerV.getDecodedTime()` is already the new functionality. 
Does it mean that you used already the release candidate from [decode_time](https://github.com/baldram/ESP_VS1053_Library/tree/decode_time) branch? 

Then it looks like it doesn't harm anything and I would merge it to master.

PS: I see that you use other registers like `SCI_AUDATA` and so on. They are already included in header file.
I wonder whether to expose each as separate API method like `getCurrentSamplerate()`, or maybe simply constants shouldn't be private? Then as public it would be accessible like this: `VS1053::SCI_AUDATA`. Then no code comments needed in your code.

```c
class VS1053 {
private:
    uint8_t cs_pin;                         // Pin where CS line is connected
    uint8_t dcs_pin;                        // Pin where DCS line is connected
    uint8_t dreq_pin;                       // Pin where DREQ line is connected
    uint8_t curvol;                         // Current volume setting 0..100%
    const uint8_t vs1053_chunk_size = 32;
    // SCI Register
    const uint8_t SCI_MODE = 0x0;
    const uint8_t SCI_STATUS = 0x1;
    const uint8_t SCI_BASS = 0x2;
    const uint8_t SCI_CLOCKF = 0x3;
    const uint8_t SCI_AUDATA = 0x5;
    const uint8_t SCI_WRAM = 0x6;
    // ...
```

fabitom, 2019-11-30:
> 
> 
> Ok, so if I see correct, `playerV.getDecodedTime()` is already the new functionality.
> Does it mean that you used already the release candidate from [decode_time](https://github.com/baldram/ESP_VS1053_Library/tree/decode_time) branch?
> 
yes, I just copy your sources to my project to src/ include/ and removed platformio version
> Then it looks like it doesn't harm anything and I would merge it to master.
>
> PS: I see that you use other registers like `SCI_AUDATA` and so on. They are already included in header file.
> I wonder [...]

yup, if users need to get some unsupported feature these registers should be public, and also read_registry, write_registry (I made slight modification and made those two methods public)

Good stuff here [https://github.com/daniel5555/Mikromedia-plus-STM32-MP3-player/blob/master/player1053.c](https://github.com/daniel5555/Mikromedia-plus-STM32-MP3-player/blob/master/player1053.c)
> 
my suggestion:
getDecodeTime (already done)
getSampleRate (read_register(SCI_AUDATA))
getChannels (read_register(SCI_AUDATA)) 
getAudioFormat (read_register(SCI_HDAT1))
getBitRate (ReadVS10xxMem(0x1e05))

baldram, 2019-12-01:
> > Does it mean that you used already the release candidate from decode_time branch?

> yes, I just copy your sources to my project to src/ include/ and removed platformio version

Ok, so I will merge the change to master and even the main goal (decoding time) will be achieved, I will keep the issue opened for a while to add support for other registers.

> my suggestion:

Thanks for suggestions!
I also compared the list you provided with the one from an [official data sheet if about "9.6 SCI Registers"](http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf). Sometimes things are not exposed directly, as the library provides kind of high-level API to simplify dealing with the chip. Below I check what is already covered directly or indirectly.
 
* 9.6.1 SCI_MODE (RW)
**Done.** 
It's already handled via: `VS1053::begin()`, `VS1053::stopSong()`, `VS1053::softReset()`.

* 9.6.2 SCI_STATUS (RW)
**Done.** via `VS1053::isChipConnected()`

* 9.6.3 SCI_BASS (RW)
**Done.** via VS1053::setTone()

* 9.6.4 SCI_CLOCKF (RW)
Low level thing **done** while executing `VS1053::begin()`

* 9.6.5 SCI_DECODE_TIME (RW)
Ok, **now we got it** thanks to you via:
`VS1053::getDecodedTime()` and `VS1053::clearDecodedTime()`

* 9.6.6 SCI_AUDATA (RW)
**TODO** as you suggested: `VS1053::getSampleRate()`

* 9.6.7 SCI_WRAM (RW)
**Done** via: `VS1053::wram_write()` and `VS1053::wram_read()`
*I don't like this lack of consequence when once methods are with camel case and once with underscore. The library is an evolution and consists a lot from parent projects. Maybe in some future version I will come up with a new names like `wramRead()` while marking underscore ones as deprecated*. 

* 9.6.8 SCI_WRAMADDR (W)
**Done**, also used in context of above `VS1053::wram_write()` and `VS1053::wram_read()`.

* 9.6.9 SCI_HDAT0 and SCI_HDAT1 (R)
Seems to be **missing**. It will require some additional investigation later on.
**TODO**: To add `getAudioFormat()` using `SCI_HDAT1` as you suggested for now. 

* 9.6.10 SCI_AIADDR (RW)
Not supported and **not needed** according to spect *Reading AIADDR is not recommended.*

* 9.6.11 SCI_VOL (RW)
**Done** via `VS1053::setVolume()`.

* 9.6.12 SCI_AICTRL[x] (RW)
Not supported at the moment. It's more for recording.

So above ones overlap also with your proposals. Now the remaining one is: `getBitRate (ReadVS10xxMem(0x1e05))`. What also looks reasonable. 

So above is kind of road map ;-)
For now, I will merge and release the library with `VS1053::getDecodedTime()` and `VS1053::clearDecodedTime()` today.

fabitom, 2019-12-01:
That will be great ;)
and the last missing feature are patches (and plugins), I don't quite understand how they work, for example flac support vs1053b need one of them [VLSI patches](http://www.vlsi.fi/en/support/software/vs10xxpatches.html) . For now I can only apply single patch by:
```c
void LoadVSPatch(void) {
  int i;
  for (i=0;i<CODE_SIZE;i++) {
    playerV.write_register(atab[i], dtab[i]);
  }
}
```
it's enought for me right now.

baldram, 2019-12-01:
Thanks for the suggestion. 
It looks like different things complement each other since patches/plugins functionality is already in TODO list for different reason. It's about MIDI support.
So it's to be implemented, one day, later on. It requires more time to understand and test it, but I'm also involved in other projects. Of course a PR is always welcome ;-)

This is really needed as it's also connected with other issue: https://github.com/baldram/ESP_VS1053_Library/issues/32. The solution is the [RTMIDI Patch](http://www.vlsi.fi/en/support/software/vs10xxpatches.html) which comes from the page you refer to.

I discovered that the [parent library code from maniacbug](https://github.com/maniacbug/VS1053) contains it already. The `VS1053.h` has `loadUserCode()`. It also has `RtMidi` class which loads RTMidi plugin. So this part needs only to be ported to ESP. Who knows, maybe even with almost no changes. I guess the code can be almost simply copied. It's just the case of testing it. The `ESP_VS1053_Library` is a fork of [Edzelf/Esp-radio](https://github.com/Edzelf/Esp-radio) from where I extracted the core driver code already ported to ESP, but some parts were missing there. For sure the goal is to extend `ESP_VS1053_Library` and add missing functionality.

> it's enought for me right now.

Yes, I also think that it would be then quite complete library.

baldram, 2019-12-01:
A [new release 1.1.2](https://github.com/baldram/ESP_VS1053_Library/releases/tag/1.1.2) is available.
It should be visible in [PlatformIO Libraries Registry](https://platformio.org/lib/show/1744/ESP_VS1053_Library) soon. 

I keep the issue opened for a while to provide other VS1053 registers handling. I mean adding: `getSampleRate()`, `getAudioFormat()` and so on. Patches/plugins support to be added separately via issue #32.

baldram, 2021-06-18:
@fabitom  
Back to the topic. We got some new features connected with it.

> and the last missing feature are patches (and plugins)

Out of the discussion [here](https://github.com/baldram/ESP_VS1053_Library/discussions/66), the plugins and patches support has been added. 
Now, adding a MIDI support for #32 should be much easier.

> yup, if users need to get some unsupported feature these registers should be public, and also read_registry, write_registry (I made slight modification and made those two methods public)

Yes, this is public now. A name of a public member is `writeRegister`.
I will add also a `readRegister` before releasing the new library version.

> `getSampleRate()`, `getAudioFormat()`, `getBitRate()`

Based on information from this comments: https://github.com/baldram/ESP_VS1053_Library/issues/41#issuecomment-559982027,  https://github.com/baldram/ESP_VS1053_Library/issues/41#issuecomment-560075584 I will open a new discussion and then it will lead to new features.

closed 2021-06-18 by baldram