baldram/ESP_VS1053_Library #44: Webradio example - increase buffer

LucasMaciuga, 2020-01-17:
Hi,

nice project.
Works great - the only thing I need to do in the Webradio example was to set the buffer to "64" instead of "32" - At 32 and 128 the sound was brassy:
[image]

[image]

Maybe it helps some one.
I also set it up with the "New ESP8266 Programmer Socket Adapter For ESPRESSIF".

Lucas

baldram, 2020-01-18:
@LucasMaciuga Thank you for sharing your observation. I think it might be a generic setting which should work better for most of (or even all) ESP8266 modules. 
The web radio example for the library provided one of the collaborators. 

Are you willing to provide a pull-request with example correction? 

Additionally, I would suggest considering a line of comment to emphasise that this is a place for experimenting with the buffer size.
Above L89 could be added something like:
```C
// The buffer size 64 seems to be optimal. At 32 and 128 the sound might be brassy.
```

baldram, 2020-01-19:
@LucasMaciuga Thank you!

closed 2020-01-19 by baldram

philippedc, 2020-01-21:
Hi, sorry I'm a bit late. I fixed the buffer to 16384 to avoid some trouble

philippedc, 2020-01-21:
I do not understand the problem : for some radio stations everything is fine. For some others, I get more or less "couik" that tired me... What I have noticed is that when there is the "couik" - sorry I do not know the term in English ...- I do not loose the audio message : I can suppose then it is a problem of speed, even if it is working fine for always the same stations.
May be to insert a rotational buffer will solve the problem, but I do not know how to do - I'm not good in programming -
neither how to test to decrease the number of bit resolution in poor speed conditions.
I have made many tests about the buffer size. I did not heard any difference between 32, 64 or 128 bits. But when I've tried big larger size, until the maximum possible with ESP8266, eg 16384, many "couik" disappeared for middle quality stations. 
I have also taken off the "if( client.available() > 0 )" test, as this does not help a lot.
I have also added a delay to the "if( !client.connected() )" test, like it is perform only 1 time every 5, to make possible to change the station. Otherwise the web server part that manages the web radio does not operate anymore.

fabitom, 2020-01-21:
32 bytes is safe buffer. When you check DREQ you can get only information if there is room for at least 32 bytes.
From my tests vs1053 can easy handle 32 bytes in most cases.
The problem is in client connection speed,bitrate data, cpu.

baldram, 2020-01-21:
> I fixed the buffer to 16384 to avoid some trouble

I think people will this find this discussion helpful. Some of them will some hints while experimenting with values like the one mentioned by @philippedc .

> 32 bytes is safe buffer

@fabitom @LucasMaciuga 
I wonder whether to revert this change and set this "safe buffer" 32 set initially (instead of 64)?
The code comment could stay there with a suggestion to experiment with different values.

What do you think?

fabitom, 2020-01-21:
I'll write simple example based on ring buffer and checking DREQ signal.

baldram, 2020-01-21:
> I'll write simple example based on ring buffer and checking DREQ signal.

@fabitom 
It sounds great! The ring buffer solves an issue in the right way.
If you could add such an additional example, that would be amazing.

fabitom, 2020-01-21:
ok, Can someone test this for me? I can't right now. It based on Edzelf esp-radio.
````ruby
#include <Arduino.h>
#include <VS1053.h>
#include <WiFi.h>

#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4

#define VOLUME  80

#define VS_BUFF_SIZE 32
#define RING_BUF_SIZE 20000

VS1053 player(VS1053_CS, VS1053_DCS, VS1053_DREQ);
WiFiClient client;

const char* ssid = "TP-Link";
const char* password = "xxxxxxxx";
     
const char *host = "comet.shoutca.st";
const char *path = "/1";
int httpPort = 8563;

uint16_t rcount = 0;
uint8_t* ringbuf;
uint16_t rbwindex = 0;
uint8_t* mp3buff;
uint16_t rbrindex = RING_BUF_SIZE - 1;

inline bool ringspace() {
  return ( rcount < RING_BUF_SIZE );     
}

void putring(uint8_t b ) {
  *(ringbuf + rbwindex) = b;
  if ( ++rbwindex == RING_BUF_SIZE ) {
    rbwindex = 0;
  }
  rcount++;
}
uint8_t getring() {
  if ( ++rbrindex == RING_BUF_SIZE ) {
    rbrindex = 0;
  }
  rcount--;
  return *(ringbuf + rbrindex);
}

void playRing(uint8_t b) {
  static int bufcnt = 0;
  mp3buff[bufcnt++] = b;
  if (bufcnt == sizeof(mp3buff)) {
    player.playChunk(mp3buff, bufcnt); 
    bufcnt = 0; 
  }  
}	

void setup() {
    Serial.begin(115200);
    mp3buff = (uint8_t *) malloc (VS_BUFF_SIZE);	
    SPI.begin();
    player.begin();
    player.switchToMp3Mode();
    player.setVolume(VOLUME);
    WiFi.begin(ssid, password);      
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
    if (!client.connect(host, httpPort)) {
      Serial.println("Connection failed");
      return;
    }    
    client.print(String("GET ") + path + " HTTP/1.1\r\n" +
                  "Host: " + host + "\r\n" + 
                  "Connection: close\r\n\r\n");
}

void loop() {
	uint32_t maxfilechunk;
	
    if(!client.connected()){
      Serial.println("Reconnecting...");
      if(client.connect(host, httpPort)){
        client.print(String("GET ") + path + " HTTP/1.1\r\n" +
                  "Host: " + host + "\r\n" + 
                  "Connection: close\r\n\r\n");
      }
    }

    maxfilechunk = client.available();
    if(maxfilechunk > 0){
      if ( maxfilechunk > 1024 ) {
        maxfilechunk = 1024;
      }		
      while ( ringspace() && maxfilechunk-- ) {
        putring(client.read());
      }
      while (rcount && (player.data_request())) {
        playRing(getring());
      }	  
    }
}
````

philippedc, 2020-01-21:
@fabitom it compiles but then continuously reboots after watchdog timeout

philippedc, 2020-01-21:
> The problem is in client connection speed,bitrate data, cpu.

in fact when I compare the same url radio from a PC and the web radio, the web radio is playing before the PC, and the time lap increases slowly from few mili-seconds to several seconds, until the web radio aborts

fabitom, 2020-01-22:
@philippedc ESP without external antenna will not get same signal strength. Did you change wifi function for your esp model? My sketch based on esp32, for esp8266 you need #include <ESP8266WiFi.h> instead of #include <WiFi.h>

fabitom, 2020-01-22:
Oh, I see problem :) No memory for ringbuff :)
````c
ringbuf = (uint8_t *) malloc (RING_BUF_SIZE);
````
please add malloc for ringbuff
````c
...
    Serial.begin(115200);
    mp3buff = (uint8_t *) malloc (VS_BUFF_SIZE);
    ringbuf = (uint8_t *) malloc (RING_BUF_SIZE);
    SPI.begin();
    player.begin();
...
````

philippedc, 2020-01-23:
@fabitom wow it runs now. I’m very impressed with this kind of code that I don’t know how to use, if you know a web site that explains step by step what you did I would be glad!

However..... :(   this doesn't give any help for these radio stations that give many "couiks".  Instead it seems the web radio aborts faster : after about half a minute the sound definitively stops. Even if the web server part is still available. 

The web radio I'm testing and my PC are 3 meters away from the wifi provider. The family web radio is 7 meters away the wifi provider. as both web radios gives identical quality result, I think there is not a problem of antenna.
But, when I use the PC as a reference of  the radio station listenning, I cannot explain why the web radio takes more and more advance, it looks like to go back in time. Does the "couiks" are the audible witness of these jump of time, or attempts to go back ???

As soon as I'll get some time I will test the web radio with a ESP-WROOM-32 I've just received, may be the "couiks" will disappear with a faster CPU ?

philippedc, 2020-01-23:
This very poor modification dos not stop the "couiks" but prevent against sound aborts after half a minute:

if( !client.connected() && tempo % 5 == 0 ) {
    Serial.println("Tentative de reconnexion...");
    if( client.connect(urlStation, portStation.toInt()) ) {
      client.print(String("GET ") + pathStation + " HTTP/1.1\r\n" +
                  "Host: " + urlStation + "\r\n" + 
                  "Connection: close\r\n\r\n");
     }
     return;
   }  // end of !client.connected() test
 
// si tout va bien :)
    uint32_t maxfilechunk = client.available();
    //if(maxfilechunk > 0){
      if ( maxfilechunk > 1024 ) maxfilechunk = 1024;
      while ( ringspace() && maxfilechunk-- ) putring(client.read());
      while (rcount && (player.data_request())) playRing(getring());
    //}
}    // end of loop

fabitom, 2020-01-24:
Please test with that loop:
````c
void loop() {
  if(!client.connected()){
    Serial.println("Reconnecting...");
    if(client.connect(host, httpPort)){
      client.print(String("GET ") + path + " HTTP/1.1\r\n" +
                "Host: " + host + "\r\n" + 
		"Icy-MetaData: 0\r\n" +
                "Connection: close\r\n\r\n");
    }
  }

  uint32_t maxfilechunk = client.available();
  if(maxfilechunk > 0){
    if ( maxfilechunk > 1024 ) {
      maxfilechunk = 1024;
    }		
    while ( ringspace() && maxfilechunk-- ) {
      putring(client.read());
    }
    while (rcount) {
      while (!player.data_request()) {
        delay(1);
      }	  
      playRing(getring());
    }	  
  }
}
````
and try to adjust buffer
````c
#define VS_BUFF_SIZE 32 //32,64,128
#define RING_BUF_SIZE 20000 // 12000 18000 25000
````

philippedc, 2020-01-25:
It is not better.... I think I will definitively leave some 128K internet radios.....

baldram, 2020-01-25:
>  I think I will definitively leave some 128K internet radios

@philippedc How does it work for 128kbps internet radio? 
I think it's a reasonable limitation according to such a basic implementation, even including ring buffer.
Is it still better than the simplest one without a buffer at all?

fabitom, 2020-01-25:
> It is not better.... I think I will definitively leave some 128K internet radios.....

@philippedc so 64kbps works ok ? then you should definitely test your network connection, first [ESP8266Ping](https://github.com/dancol90/ESP8266Ping)

philippedc, 2020-01-25:
The (my) web radio receiver works fine with 64k streamings, and most of 128k. 
It is not a problem of network bandwidth because I have 3 of this radio at home that can play at the same time, plus pc, plus tv... Unfortunately for some stations the sound quality is very noisy, and after a moment more or less long the receiver abort, I must reboot the device. Each time it is due to the same station, foe instance CALM classical piano. Of course I can listen many other radios, but.... 
It must be a solution, it is the vs1053 that aborts, not the esp: it never reboot by himself. 
Instead of wasting cpu time by testing if there is something in the buffer, I think it would be better to test if vs1053 has correctly decoded it.

baldram, 2020-01-31:
> I think it would be better to test if vs1053 has correctly decoded it.

@philippedc What is the stream which aborts? 
Is it that some URLs are correct and some lead to the issue? 
Are you able to provide example URLs having a problem?

philippedc, 2020-01-31:
here is some radios examples:
these work well : 
136.0.17.26:8000/stream/;.m3u
tsfjazz.ice.infomaniak.ch//tsfjazz-high.mp3
icecast.radiofrance.fr/franceculture-lofi.mp3 => OK 64K

but theseare very nosy : 
23.82.11.89:30228/stream 
23.82.11.88:5128/stream

These abort the webradio after few minutes of playing:
icecast.radiofrance.fr/franceculture-midfi.mp3
icecast.radiofrance.fr/francemusique-midfi.mp3

baldram, 2020-02-01:
Thank you for your links.

> do you mean the version of your library ? It is the actual version.

The actual one is 1.1.2 and has only a few new methods. I think it's not the library version issue. 
If you build more advanced radio, people buffer the incoming encoded audio stream in an external SPI RAM. So the ring buffer using internal memory is a step to improve the quality, but still not enough. Here might be a problem.
If about the library itself, the core has not been changed for years. 
The library is extracted from [Esp-radio](https://github.com/Edzelf/Esp-radio) (the core VS1053 driver code only).
The original one (before porting it to ESP) was implemented around eight years ago, see here: [maniacbug/VS1053](https://github.com/maniacbug/VS1053). It's still very close. So it's matured and stable code.

Are you able to attach the current version of the radio code you built out of the discussion in this thread? 

I need to find the box with my electronics stuff, and then I would try to reproduce the issue on my side.

philippedc, 2020-02-01:
[WebRadio-v4.25.ino.txt](https://github.com/baldram/ESP_VS1053_Library/files/4142397/WebRadio-v4.25.ino.txt)
[WebRadio-v5.1.ino.txt](https://github.com/baldram/ESP_VS1053_Library/files/4142399/WebRadio-v5.1.ino.txt)

baldram, 2020-02-01:
Thanks for code. 
Actually, the web radio example attached to the library is by @vincegellar :)
Pull-requests are very welcome. This example is one of them.

I will try to find some spare time to test a similar ring buffer radio version on my side. The goal is to observe whether the problem is reproducible in my network. 
I see you run it with ESP8266. I will try with ESP32, which should be more performant.