baldram/ESP_VS1053_Library #46: esp32 platformio build error

nopnop2002, 2020-03-08:
Hello.
I try to build firmware.
But there is some error.
Do you have any help??

```
$ pio run -e esp32dev
Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
---------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30300.190916 (3.3.0), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio/build/esp32dev/src/VS1053.o
In file included from src/VS1053.cpp:33:0:
src/VS1053.h:37:21: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://platformio.org/lib/search?query=header:Arduino.h
*
*****************************************************************

compilation terminated.
Compiling .pio/build/esp32dev/bootloader/bootloader_start.o
In file included from /home/nop/.platformio/packages/framework-espidf/components/bootloader/subproject/main/bootloader_start.c:18:0:
/home/nop/.platformio/packages/framework-espidf/components/log/include/esp_log.h:20:23: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://platformio.org/lib/search?query=header:sdkconfig.h
*
*******************************************************************

compilation terminated.
Compiling .pio/build/esp32dev/bootloader/bootloader_support/src/bootloader_clock.o
In file included from /home/nop/.platformio/packages/framework-espidf/components/esp32/include/rom/uart.h:19:0,
                 from /home/nop/.platformio/packages/framework-espidf/components/bootloader_support/src/bootloader_clock.c:14:
/home/nop/.platformio/packages/framework-espidf/components/esp32/include/esp_attr.h:17:23: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://platformio.org/lib/search?query=header:sdkconfig.h
*
*******************************************************************

compilation terminated.
*** [.pio/build/esp32dev/src/VS1053.o] Error 1
*** [.pio/build/esp32dev/bootloader/bootloader_start.o] Error 1
Compiling .pio/build/esp32dev/bootloader/bootloader_support/src/bootloader_common.o
*** [.pio/build/esp32dev/bootloader/bootloader_support/src/bootloader_clock.o] Error 1
In file included from /home/nop/.platformio/packages/framework-espidf/components/bootloader_support/src/bootloader_common.c:15:0:
/home/nop/.platformio/packages/framework-espidf/components/newlib/platform_include/assert.h:20:23: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://platformio.org/lib/search?query=header:sdkconfig.h
*
*******************************************************************

compilation terminated.
*** [.pio/build/esp32dev/bootloader/bootloader_support/src/bootloader_common.o] Error 1
================================================== [FAILED] Took 1.38 seconds ==================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      IGNORED
esp32dev       FAILED    00:00:01.376
============================================= 1 failed, 0 succeeded in 00:00:01.376 =============================================

```

My environment:
```
$ pio update -c
Checking tool-scons                      @ 2.20501.191222 [Up-to-date]

Platform Manager
================
Platform Espressif 32
--------
Checking espressif32                     @ 1.11.2         [Up-to-date]
Checking tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Checking toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Checking framework-arduinoespressif32    @ 3.10004.200129 [Up-to-date]
Checking framework-espidf                @ 3.30300.190916 [Up-to-date]
Checking tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 32 (Stage)
--------
Checking espressif32                     @ cff56ff        [Up-to-date]
Checking tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Checking toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Checking framework-espidf                @ 3.30300.190916 [Up-to-date]
Checking tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 8266
--------
Checking espressif8266                   @ 2.3.3          [Up-to-date]
Checking tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
Checking tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Checking tool-esptool                    @ 1.413.0        [Up-to-date]
Checking framework-arduinoespressif8266  @ 3.20603.200130 [Up-to-date]
Checking toolchain-xtensa                @ 2.40802.191122 [Up-to-date]

Platform ST STM32
--------
Checking ststm32                         @ 6.0.0          [Up-to-date]
Checking tool-stm32duino                 @ 1.0.1          [Up-to-date]
Checking tool-dfuutil                    @ 1.9.190708     [Up-to-date]
Checking framework-arduinoststm32        @ 3.10700.191028 [Up-to-date]
Checking toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Checking framework-arduinoststm32-maple  @ 1.10000.190819 [Up-to-date]
Checking tool-openocd                    @ 2.1000.190707  [Up-to-date]

Library Manager
===============
Library Storage: /home/nop/.platformio/lib
Checking STM32Ethernet                   @ 1.0.5          [Up-to-date]
Checking STM32duino FreeRTOS             @ 10.0.1         [Up-to-date]
Checking STM32duino LwIP                 @ 2.1.2          [Up-to-date]
```

nopnop2002, 2020-03-08:
Sorry.
This is my misstake.

closed 2020-03-08 by nopnop2002

baldram, 2020-03-09:
Good luck!