baldram/ESP_VS1053_Library #51: Set volume for left and right channels

fabitom, 2020-03-30:
For my project I need balance volume on both channels.
My changes of setVolume() function for that:
````
//.h delcaration
    void setVolume(uint8_t volL, uint8_t volR = 101);
````
````
//.c
void VS1053::setVolume(uint8_t volL, uint8_t volR) {
    uint16_t value, valueL, valueR;

    curvol = volL;
    valueL = map(volL, 0, 100, 0xFF, 0x00);
    if (volR < 101) {
       valueR = map(volR, 0, 100, 0xFF, 0x00);
       value = (valueL << 8) | valueR;
    } else {
       value = (valueL << 8) | valueL; 
    }   
    write_register(SCI_VOL, value);
}
````
default value to save backward compatible.  The problem is with getVolume() which return one value.
Maybe some balance function would be better?

baldram, 2020-03-30:
Thanks for this issue. It looks like a valuable extension.
You are right. The backward compatibility is essential. 

> The problem is with getVolume() which return one value.
Maybe some balance function would be better?

Yeah, good question: how to to find the best compromise to extend it while keeping it unchanged?
I had a look at the original library source from [maniacbug](https://github.com/maniacbug/VS1053/blob/master/VS1053.cpp) (before Edzelf modified it) and there was no `getVolume()` and no problem :joy: 
I see that Adafruit has `setVolume()` with two arguments, but also no getter.

The question is whether reading the Left and Right values are needed?
Then maybe the mentioned balance-related function would be a good idea. If it existed without conflicts with a volume one, it would be perfect. So for those who need this simple interface, a regular volume function would be a choice. But for more precise dealing, the balance would be an answer.
After the balance modification, the recalculated volume (the current value) would be provided by the regular `getVolume()` method. Make sense.

fabitom, 2020-03-31:
> Then maybe the mentioned balance-related function would be a good idea. If it existed without conflicts with a volume one, it would be perfect. So for those who need this simple interface, a regular volume function would be a choice. But for more precise dealing, the balance would be an answer.

I misunderstood the volume setting concept in the previous example. Separate balance function in pull request.
[pull 53](https://github.com/baldram/ESP_VS1053_Library/pull/53)

baldram, 2020-03-31:
@fabitom What do you think? Maybe if retrieving left and right balance value (I mean a getter for it) is not needed, then exactly what you proposed would be sufficient? 
So just an additional  `setVolume` method with two parameters? The `getVolume` would stay unchanged. So as it is now. Would you like to provide a PR?

baldram, 2020-03-31:
Thank you for the PR!

closed 2020-03-31 by baldram