baldram/ESP_VS1053_Library #55: stream local content

trombose009, 2020-05-21:
Hello, 

first of all, I know the original edzelf esp-radio. But I am not able to go through the code and understand it. 
Your two examples are working well. Especially the webradio example. 

My question is: Does anyone know an example where it is shown how you can stream oder just play content from your own LAN? for example with such a streaming address:
\\192.168.1.1\mediaserver\music\title01.mp3

greetings
SVen

baldram, 2020-05-22:
Hi @trombose009, and thank you for using the library. Please don't forget to leave a star if you like it ;-)

This question is a little bit not connected with the VS1053 library itself as its task is to send data to the device. How the data is getting, it's the other case. In esp-radion it's happening probably in `connecttofile()`. I googled for some article with examples [here](https://tttapa.github.io/ESP8266/Chap11%20-%20SPIFFS.html), but I'm not sure whether `SPIFFS.open()` supports what you need. But for instance, if you create a local FTP server, I see some randomly googled solution [here](https://diyprojects.io/esp8266-ftp-server-spiffs-file-exchange-rapid-development-web-server/).

I don't have any ready idea for you, and maybe someone else can help here?

What I could advise, I would look at examples of implementing "Samba client" or "SMB client" for ESP devices.
I'm guessing only, seeing this `\` in the path you showed, that we talk about the local Windows server or at least Samba server. 

Does it help in any way?

Regards, Marcin

baldram, 2020-06-24:
No response from @trombose009 . 
I'm closing the issue. It's connected in general with Arduino programming.
Hope the hints were useful.

closed 2020-06-24 by baldram