baldram/ESP_VS1053_Library #56: Can't read SCI registers?

mattfromeurope, 2020-09-12:
Hi,

I'm trying to get this library to work. I can compile the Web Radio test program, but when I run it, the debug output shows:
`testComm(): VS1053 error retry SB:012C R1:0000 R2:0000`
And similar. And of course, there's no sound whatsoever.

It seems like the ESP32 can't read the SCI registers via SPI - or did I understand it wrong?

I wired it exactly as put in the example. I'm getting pretty iffy, as I can't use the MP3 board!

My hardware:

- [AZDelivery ESP32 WROOM32](https://www.az-delivery.de/products/esp32-developmentboard)
- [LC Technology VS1003/1053 MP3 Codec](https://www.amazon.de/LC-Technologie-VS1053-MP3-Decoder-Mikrofon/dp/B077KB25T5)

Anything I can do on my side to get it working?

baldram, 2020-09-13:
Hi, thank you for trying. It have taken a while since I played with this stuff. Similar issues I had if the wiring was incorrect. Even GND cable is important and of course other while properly defined in the program. I would recommend to double check the wiring.
I thing it's a good idea to start with the simplest hello world program from examples, the mp3 player. Then radio if any sound works.

Good luck

mattfromeurope, 2020-09-13:
OK. I redid the wiring, and now I don't get any error messages anymore - at least in the mp3 player demo. However, I still can't hear sound. Any ideas what I could have done wrong?

baldram, 2020-09-13:
Good to hear. It still sounds like some hardware issue. I remember that correct wiring was very important, as not always reported with errors. Sometimes I had a problem like the correctly wired board stopped working (the same setup was earlier ok, no modifications later). The problem was that one of the cables was not conducting, it was disconnecting sometimes. 

I experimented with different boards. The test report is here: #1 . Even no errors I would still suggest to double-check looking at different set ups from mentioned link. The smallest details are important here.

Fingers crossed 🤞

baldram, 2020-10-04:
Closing issue for inactivity. Solution suggestions above.

closed 2020-10-04 by baldram