baldram/ESP_VS1053_Library #57: How to play .wav files?

A-R-S-D, 2020-11-26:
Hi, Thanks for this really cool library, i have a question about playing WAV format, a searched the documentation and this is mentioned that switchedToMp3() function is for switching from MIDI mode to MP3 mode, so is there any single mode for WAV format too?
And does this currently available API do the job of playing WAV formats too?
Thanks so much.

baldram, 2020-11-29:
Hi, thanks for the good words and also please do not forget to star the project ;-)

The library is just a driver that allows you to deal with your hardware according to the specification http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf . If ESP device memory limitation is not a problem and for some reason, you prefer using WAVE over MP3, it's your choice. 
The "switch to mp3" is in fact switching between MIDI and standard audio mode. So this is also for WAV. 
You can find even in git history the change, where mistakenly the demo file was a WAV actually: https://github.com/baldram/ESP_VS1053_Library/commit/369cbff71c94e4838c409543e12be514a01894ba .
The size got reduced after switching to mp3, but it worked fine with the previous file.

Is this clarification helpful?
Best regards and good luck.

A-R-S-D, 2020-11-30:
Thanks a lot for kind response. Yes, That was actually what i meant. 

But i another question appeared to me as you said "device memory limitation". I am just trying to learn audio processing, So sorry if my words doesn't make sense to you. but why memory limitation? we can load WAVE file chunk by chunk from a SD Card and play one chunk while fetching the next chunk. So does the memory limitation really matters?

Thanks again.

baldram, 2020-11-30:
With SD card it's ok then. Always mp3 lets you put more, but the card gives you enough space. 

Good luck 👍

A-R-S-D, 2020-12-01:
Yes i think so, Thanks.

Regards. 🌹

closed 2020-12-01 by A-R-S-D