baldram/ESP_VS1053_Library #62: Errors on begin

xtrinch, 2021-02-08:
I get the following errors from the begin sequence:

```
[D][VS1053.cpp:156] begin(): 

[D][VS1053.cpp:157] begin(): Reset VS1053...

[D][VS1053.cpp:161] begin(): End reset VS1053...

[D][VS1053.cpp:132] testComm(): Slow SPI,Testing VS1053 read/write registers...

[D][VS1053.cpp:132] testComm(): Fast SPI, Testing VS1053 read/write registers again...

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8001 R1:8000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8004 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8007 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:800A R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:800D R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8010 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8013 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8016 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8019 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:801C R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:801F R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8022 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8025 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8028 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:802B R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:802E R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8031 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8034 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:8037 R1:0000 R2:0000

[D][VS1053.cpp:140] testComm(): VS1053 error retry SB:803A R1:0000 R2:0000

```

But my radio works great & the sound is great, so what is this about? I get these errors on every boot. Can I opt out of these "tests", if they're not really necessary?

Dr-Dawg, 2021-02-11:
I would not recommend to comment the line

```
    testComm("Fast SPI, Testing VS1053 read/write registers again...\n");
```

in VS1053.cpp, but rather try to find the reason for the fast SPI error messages.
Since testComm uses the Volume setting for the test,
have you tried changing the Volume per player.setVolume(VOLUME) in your main loop or  during the setup?

xtrinch, 2021-02-12:
Yes, I have, it works fine.

baldram, 2021-02-21:
> Yes, I have, it works fine.

@xtrinch  Can I close the issue?

xtrinch, 2021-02-21:
Well, sure, its your lib, but the testComm function still doesnt work for me.

fabitom, 2021-02-21:
@xtrinch testComm errors means that you have some SPI problems and your module works with slow SPI (200kHz) insteaad of fast SPI 4MHz.

xtrinch, 2021-02-22:
What could be the possible causes of that?

baldram, 2021-06-12:
> What could be the possible causes of that?

Those are like hardware or wiring issues. Had similar things in the past I remember because of wiring. 
Only guessing that it might be a broken cable. The `testComm` warns about the issue that something is messed up around SPI.

closed 2021-06-12 by baldram