baldram/ESP_VS1053_Library #64: Adjust data rate

h1aji, 2021-05-20:
from Edzelf`s code:

`void     AdjustRate ( long ppm2 ) ;                  // Fine tune the datarate`

```void VS1053::AdjustRate ( long ppm2 )                  // Fine tune the data rate 
{
  write_register ( SCI_WRAMADDR, 0x1e07 ) ;
  write_register ( SCI_WRAM,     ppm2 ) ;
  write_register ( SCI_WRAM,     ppm2 >> 16 ) ;
  // oldClock4KHz = 0 forces  adjustment calculation when rate checked.
  write_register ( SCI_WRAMADDR, 0x5b1c ) ;
  write_register ( SCI_WRAM,     0 ) ;
  // Write to AUDATA or CLOCKF checks rate and recalculates adjustment.
  write_register ( SCI_AUDATA,   read_register ( SCI_AUDATA ) ) ;
}

baldram, 2021-05-20:
I have currently no possibility to make an adjustment and test this with a device. If this is something urgent I would consider creating a MR.  

One more note. I see that in Edzelf's code this is `void AdjustRate ( long ppm2 )` but it looks like a name is by mistake like this as it doesn't meet convention. It shall be rather `void adjustRate ( long ppm2 )` .

Open for MRs. Or few weeks later I will try with it.

h1aji, 2021-05-22:
I have got esp8266 and SRAM , just thought to complete setup and try to improve performance.
Not urgent, I have this change added locally and it works just fine

h1aji, 2021-06-03:
Hi @baldram,
I found it extremely hard to create a MR as your fork is very different, I tried to create MR from orphan branch but it cannot see the previous commits and instead re-creates existing files.
Anyway, I have attached two files that required change. Can you please commit?
Thank you
[vs1053.zip](https://github.com/baldram/ESP_VS1053_Library/files/6588258/vs1053.zip)

baldram, 2021-06-03:
Thanks. Is it still about adding the Adjust Rate piece of code? L In general, the codebase should be still very similar, some cleanup tasks were intentionally postponed.
All right, I would try to add it. 

It would be great if you would be able to use the enhanced version of ESP_VS1053 as Platform IO dependency in your project. Is it possible?

h1aji, 2021-06-04:
> Is it still about adding the Adjust Rate piece of code?
Yeah, I just pulled latest code and added adjustRate piece, you can see it in diff

baldram, 2021-06-04:
All right, I will add it later today or during the weekend. 
I will be great if you'll be able to test the updated PlatformIO dependency in your project.
We will have the feature confirmed that works as expected.
Thanks.

baldram, 2021-06-12:
Hi @h1aji , sorry it took a while, but I had a very busy time at work doing some over-hours. 
Couldn't do it earlier. 

However, it has been just delivered!
As you wished, there is a new `adjustRate` public method.

Are you willing to test a snapshot from Github directly in your project? Before it's released in PlatformIO.

Supported schemes:
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html#version-control

h1aji, 2021-06-13:
Thank you.

> Are you willing to test a snapshot from Github directly in your project?

Yes, I will be pulling latest code from master branch.

closed 2021-06-13 by h1aji