baldram/ESP_VS1053_Library #74: Could there be a new release wrt the firmware patches?

CelliesProjects, 2021-08-04:
Hi baldram. 
People installing this lib from the Arduino IDE can not use the latest commits since they are stuck on 1.1.2.
To get the latest commits you have to use git on the command-line which is a barrier to a lot of users.
See https://github.com/CelliesProjects/ESP32_VS1053_Stream/issues/3

CelliesProjects, 2021-08-04:
Wrong repo. Closing.

baldram, 2021-08-04:
Hey, yes. I haven't got any feedback whether all is fine. It means, yes. Then I will release a current state. A new release is on the way. Tomorrow, as sailing now. 😄 Best regards, Marcin

baldram, 2021-08-07:
@CelliesProjects It's released. https://github.com/baldram/ESP_VS1053_Library/releases/tag/1.1.4
Out of my mistake it is 1.1.4 what actually make sense after long time.
Would you please have a look whether all fine from Arduino IDE now? I have no experience at all if about this stack. 

I will check how about PlatformIO repository, whether it was correctly published there.

CelliesProjects, 2021-08-09:
@baldram Issue has been solved. Thanks!

closed 2021-08-09 by CelliesProjects

baldram, 2021-08-09:
All right, also here a new release is visible: https://platformio.org/lib/show/1744/ESP_VS1053_Library 👌