baldram/ESP_VS1053_Library #95: Bass / treeble change

jackjean88, 2022-03-25:
Hi all,
My Web radio is now running well, with a really good sound thru headphones at a 192kbps stream. 
Before integrating a 2.8" Touch screen, I just test it on a 2.1 loud speaker sound set.
I feel a little upset due to a weak bass level, compared with my existing streamer that runs on the same set.

Just digging into the VS1053.cpp file, I found these lines :

`void VS1053::setTone(uint8_t *rtone) { // Set bass/treble (4 nibbles)
    // Set tone characteristics.  See documentation for the 4 nibbles.
    uint16_t value = 0; // Value to send to SCI_BASS
    int i;              // Loop control

    for (i = 0; i < 4; i++) {
        value = (value << 4) | rtone[i]; // Shift next nibble in
    }
    writeRegister(SCI_BASS, value); // Volume left and right
}`

This proves that a tone setup is possible, but I do not find any example showing to include it into the IDE sketch...

Does somebody able to help me in that way ?

Dr-Dawg, 2022-03-25:
I understand the code and the VS1053 datasheet like sending an array of four 4-Bit values (i.e. array of four uint8_t values filled with values ranging from 0-15) to the setTone function.
The order of the values is ST_AMPLITUDE, ST_FREQLIMIT, SB_AMPLITUDE, SB_FREQLIMIT,
as described in the datasheet, chapter 9.6.3:
ST_AMPLITUDE, Bits  15:12 Treble Control in 1.5 dB steps (-8..7, 0 = off)
ST_FREQLIMIT, Bits 11:8 Lower limit frequency in 1000 Hz steps (1..15)
SB_AMPLITUDE, Bits 7:4 Bass Enhancement in 1 dB steps (0..15, 0 = off)
SB_FREQLIMIT, Bits 3:0 Lower limit frequency in 10 Hz steps (2..15)

An example for the use of player.setTone can be found in the code of @Edzelf , from which this library is forked:
https://github.com/Edzelf/Esp-radio

Note that the datasheet reports a possible bug in volume control, when bass enhancer or treble control are active.
Appearantly you need to load the default patch, which is currently not handled correct in the examples, see #89.

jackjean88, 2022-03-25:
@Dr-Dawg 
Thank you for this information.
I must admit to having a terrible headache trying to figure out the Esp_radio.ino code.
My level is largely insufficient for the manipulation of arrays and even worse for the whole sketch! 
Of course I understand that it is easier to play with a numerical value to express a volume than to define 4 values (2 amplitude + 2 frequency).
I would have liked to treat this setting like my current project which acts on the volume by entering a character in the serial monitor (current: + / - ).
We could imagine entering a string like "T2 F10" for
"ST_AMPLITUDE 2xdB and ST_FREQLIMIT 10x1000 Hz
and "B12 F3" for the bass.
The conversion of this received string into arrays values should be within my reach. However, once these commands have been translated into arrays, I still have to include them in the SetTone function.

Concerning the default patch I understand that I have to add the command "player.loadDefaultVs1053Patches();" after the line "player.switchToMp3Mode();" but where are these DefaultVs1053 patches?

jackjean88, 2022-03-26:
Sorry for the brief moment of panic. Thanks to your information (at 70 years old) I have simultaneously made a big step in my coding practice and in the control of the tone of my project.
I got the expected result, without suffering from the announced bug, so no need for a patch.

Dr-Dawg, 2022-03-27:
Good to hear, I was about to prepare some example code, but didn't find the time yet. So it sounds like this obsolete now :-)

baldram, 2022-03-28:
@jackjean88 Good to hear all worked. 
Maybe you could share here some code snipped what worked well on your side with a few words of comment?
This might be helpful for others with a similar problem in the future when browsing this issue.

jackjean88, 2022-03-28:
It remains my intention to share extracts of my code, once it has met all the requirements of my personal specifications.
Why extracts? Because I still have a big part to deal with: a Nextion touch screen control.
I'm not going to wait for this part before sharing. I have however a last audio point to solve: still on the theme of the tone, we cannot repeat the definition of the array rtoneCustom[4]. However, this array only allows to act on two points (one treble + one bass) of the equalization curve. I would like to be able to act on several points, a bit like a graphic equalizer.
Any ideas?

baldram, 2022-03-28:
> It remains my intention to share extracts of my code, once it has met all the requirements of my personal specifications.
Why extracts? Because I still have a big part to deal with: a Nextion touch screen control.

Yes, I meant only short snippets like one or few lines of code where it's about executing `setTone` and which example parameters worked. That's it. It's not about the entire code of your application with screen handling and so on 😄  But yeah, when audio troubles are solved. Good luck 👍 
Unfortunately, I haven't played with graphics eq to share any ideas from my side.

jackjean88, 2022-03-28:
My project started from the project described here:
https://electroniqueamateur.blogspot.com/2021/03/esp32-et-vs1053-ecouter-la-radio-sur.html
It is free of all touch and button controls. Actions are simply caused by typing characters into the serial monitor: Increase / decrease volume by + / - and browse the radio station list by the "n" key.

The differences with the example are as follows:
1 - buffer size change from 64 to 128
2 - the URLs of the stations were not working, here are the 7 I chose:
      strcpy(host, "icecast.radiofrance.fr");
      strcpy(path, "/franceinter-hifi.aac");
      httpPort = 80;

      strcpy(host, "icecast.radiofrance.fr");
      strcpy(path, "/fbsudlorraine-midfi.mp3");
      httpPort = 80;

      strcpy(host, "icecast.radiofrance.fr");
      strcpy(path, "/franceinfo-hifi.aac");
      httpPort = 80;

      strcpy(host, "icecast.radiofrance.fr");
      strcpy(path, "/francemusique-hifi.aac");
      httpPort = 80;

      strcpy(host, "stream.rfm.fr");
      strcpy(path, "/rfm-wr7.mp3");
      httpPort = 8000;

      strcpy(host, "mediaserv38.live-streams.nl");
      strcpy(path, "/live");
      httpPort = 8006;

3 - On line 95, I changed "HTTP/1.1" to
                                         **"HTTP/1.0"** 
      to disallow the processing of non-audio streams.
4 - In the following lines, I inserted the disabling of non-audio data:
           "Host: " + host + "\r\n" +
           **"Icy-MetaData: 0\r\n" +**
            "Connection: close");
5 - In the Setup I inserted the following lines in bold :
        player.begin();
        player.switchToMp3Mode();
        player.setVolume(volume);
        **uint8_t rtoneCustom[4] = { 7, 15, 13, 8 } ;
        player.setTone(rtoneCustom);**

Choice of cut-off and frequency values.

They depend on the amplifier and speaker part. My 2.1 system (i.e. with subwoofer). The 2 small satellites are very precise in the highs and deserve to be well served.
From the table that I am distributing below, I have therefore chosen 7, 15 in the treble part: 7 for a maximum rise (10.5 dB) for frequency 15 (15 KHz).
In the same way, the subwoofer had to express itself better, so I chose 13, 8 for the bass: 13 for a significant (but not maximum) rise of 13dB out of 15 possible for frequency 8, i.e. 70Hz.
I must therefore do without a graphic equalizer (with multiple points) since the VS1053 only manages two points.
But this was enough to obtain a detailed and well rounded sound.

Hereafter the table of choice of the gain and cut-off values:
0  HA       0 Off
                1  +1.5 dB gain for treble
                2  +3.0 dB gain for treble
                3  +4.5 dB gain for treble
                4  +6.0 dB gain for treble
                5  +7.5 dB gain for treble
                6  +9.0 dB gain for treble
                7 +10.5 dB gain for treble
                8 -12.0 dB gain for treble
                9 -10.5 dB gain for treble
               10  -9.0 dB gain for treble
               11  -7.5 dB gain for treble
               12  -6.0 dB gain for treble
               13  -4.5 dB gain for treble
               14  -3.0 dB gain for treble
               15  -1.5 dB gain for treble
1  HF        1  1 kHz cutoff for treble
                2  2 kHz cutoff for treble
                3  3 kHz cutoff for treble
                4  4 kHz cutoff for treble
                5  5 kHz cutoff for treble
                6  6 kHz cutoff for treble
                7  7 kHz cutoff for treble
                8  8 kHz cutoff for treble
                9  9 kHz cutoff for treble
               10 10 kHz cutoff for treble
               11 11 kHz cutoff for treble
               12 12 kHz cutoff for treble
               13 13 kHz cutoff for treble
               14 14 kHz cutoff for treble
               15 15 kHz cutoff for treble
2  LA        0 Off
                1  +1 dB gain for bass
                2  +2 dB gain for bass
                3  +3 dB gain for bass
                4  +4 dB gain for bass
                5  +5 dB gain for bass
                6  +6 dB gain for bass
                7  +7 dB gain for bass
                8  +8 dB gain for bass
                9  +9 dB gain for bass
               10 +10 dB gain for bass
               11 +11 dB gain for bass
               12 +12 dB gain for bass
               13 +13 dB gain for bass
               14 +14 dB gain for bass
               15 +15 dB gain for bass
3  LF         2  10 Hz cutoff for bass
                3  20 Hz cutoff for bass
                4  30 Hz cutoff for bass
                5  40 Hz cutoff for bass
                6  50 Hz cutoff for bass
                7  60 Hz cutoff for bass
                8  70 Hz cutoff for bass
                9  80 Hz cutoff for bass
               10  90 Hz cutoff for bass
               11 100 Hz cutoff for bass
               12 110 Hz cutoff for bass
               13 120 Hz cutoff for bass
               14 130 Hz cutoff for bass
               15 140 Hz cutoff for bass

Dr-Dawg, 2022-03-29:
Of course, all frequencies above ST_FREQLIMIT and below SB_FREQLIMIT are affected, too..

I'm wondering whether the 15 KHz setting might be too high, a lot of people do not hear anything above that frequency. 

Furthermore there is one little correction about the bass frequencies, I think the frequency is SB_FREQLIMIT * 10 Hz . The datasheet states:
```
For example setting SCI_BASS to 0x00f6 will have 15 dB enhancement
below 60 Hz.
```
0x00f6 is 1111 0110, so  SB_AMPLITUDE = 15 and SB_FREQLIMIT = 06
That means setting rtoneCustom[3] = 8 is a bass enhancement below 80 Hz
Btw, the datasheets recommendation for SB_FREQLIMIT is:
```
SB_AMPLITUDE should be set to the user’s preferences, and SB_FREQLIMIT to roughly 1.5 times the lowest frequency the user’s
audio system can reproduce.
```