baldram/ESP_VS1053_Library #104: Streaming 256kb/s generates artifacts

postcom, 2023-03-31:
I found that streaming audio from  the internet at 256kb/s generate strange audio artifacts. Is this something know?
I am streaming from this address: http://stream-mixtape-geo.ntslive.net/mixtape, but the issues is present on every station with 256kbps stream I tried.

Codec: MPEG Audio layer 1/2
Sample rate: 44100Hz
Bits per sample: 32
Bitrate: 256kbps

Thank you for this great tool!

Dr-Dawg, 2023-04-06:
I guess you need some form of buffering like done by @fabitom / @Edzelf
https://github.com/baldram/ESP_VS1053_Library/pull/54

postcom, 2023-04-07:
So the issue was caused by chunked transfer encoding. I have no idea what that means, but this line was what made the difference:
`#define HTTPVER       "HTTP/1.0" //use HTTP/1.0 to force not chunked transfer encoding`

if someone has some pointer to help me figure out what this chunked transfer does, I'd be extremely happy to read about it.

closed 2023-04-07 by postcom