baldram/ESP_VS1053_Library #108: Question: Is it possible to use the second SPI for this lib? Using M5Stack Core2.

CelliesProjects, 2023-10-28:
I am using great this lib for a while now. 
For my current project I planned on using this library on a M5stack Core2 board with the M5Stack driver lib. 

Is it possible to use the VS103 lib with the HSPI bus in tandem with the VSPI bus?

Dr-Dawg, 2023-10-28:
I used a TFT with HSPI and the library with VSPI. This looked something like:

```
// MH VSPI CS = 5, SCLK = 18, MISO = 19, MOSI = 23 --- VS1053
// MH HSPI CS = 15, SCLK = 14, MISO = 12, MOSI/SDA = 13 --- TFT
// TFT A0  - Pin17 / TX2
// TFT Reset - Pin16 / RX2
```
(just comments to remind me which pins are used for what purpose)
and

```
#include "tft_funktionen.h"

#include <SPI.h> // obsolet, da schon woanders inkludiert?

// MH fuer zweiten SPI Kanal: HSPI
SPIClass SPI2(HSPI);

// MH TFT ESP32  
  #define TFT_CS        15 // 5 -15 CS
  #define TFT_RST       16 // Reset
  #define TFT_DC        17 // A0

// For 1.44" and 1.8" TFT with ST7735 use:
// MH - ESP32: SPI2 Übergabepbjekt wird akzeptiert (für HSPI, sonst für VSPI drei Standardübergabeparameter)
Adafruit_ST7735 tft = Adafruit_ST7735(&SPI2, TFT_CS, TFT_DC, TFT_RST);
```
and
```
SPI.begin();
```

CelliesProjects, 2023-11-02:
Hi @Dr-Dawg 
That wont work.

The lcd SPI pins on M5Stack boards are not changeable.

So I have to change to the second SPI bus for the VS1053 or look if both libraries work together and play nice on the first SPI bus. 
(I only got bad experience with M5Stack libraries working together and playing nice)

So I think I rather have a second SPI bus for the VS1053 board.

@baldram  But looking at the code, it is not possible atm ? 

Or am I missing something here?

baldram, 2023-11-02:
Honestly, I don't have experience with M5Stack. It's very interesting to see how well this library will support additional hardware. I'm not sure how successful adapting it will be since many things are hardcoded in the original code from which the fork originated.

witnessmenow, 2024-01-01:
The typical way of handing this is to allow the user to pass in a spi reference as part of constructor or begin and to default it if they don't 

That way users have a way of configuring whatever combination of spi ports and pins they want

The esp32 SD card library for example
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.h#L31

An example of using that functionality
https://github.com/witnessmenow/ESP32-Trinity/blob/master/examples/TrinityFeatures/SDCardTest/SDCardTest.ino#L210

CelliesProjects, 2024-01-03:
@all Thanks for the input.

After some trying I have a POC running. 
Uses the M5GFX library instead of the M5Stack all-in-one driver.

Use the predefined SPI pins and a M5Stack flag in `platformio.ini`:
```
build_flags =
    -D VS1053_CS_PIN=13
    -D VS1053_DCS_PIN=26
    -D VS1053_DREQ_PIN=36
    -D SPI_CLK_PIN=18
    -D SPI_MISO_PIN=19
    -D SPI_MOSI_PIN=23
    -D M5STACK_M5GFX
```

And in `blabla.ino`:

```c++
...
#if defined(M5STACK_M5GFX)
#include <M5GFX.h>
M5GFX display;
#endif

ESP32_VS1053_Stream audio;

setup()
    SPI.setHwCs(true);
    SPI.begin(SPI_CLK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN);

#if defined(M5STACK_M5GFX)
    display.init();
    display.setTextSize((float)display.width() / 160);
    display.printf("eStreamplayer \n%s\n", GIT_VERSION);

    log_i("Disabling M5Stack DAC");
    pinMode(25, OUTPUT); // 25 on Grey and Fire
    digitalWrite(25, 0);
#endif

    audio.startDecoder(VS1053_CS_PIN, VS1053_DCS_PIN, VS1053_DREQ_PIN);
    ...
    ...

```

`audio` is an object controlling the VS1053 hw. 
Now the codec chip and the M5Stack display can share the SPI bus.

baldram, 2024-01-03:
Hey @CelliesProjects if I'm correct you've tested the M5Stack, right? Maybe we can add a not here: https://github.com/baldram/ESP_VS1053_Library/issues/1 with a reference to this issue (https://github.com/baldram/ESP_VS1053_Library/issues/108) and it's last (https://github.com/baldram/ESP_VS1053_Library/issues/108#issuecomment-1875305006) comment?

CelliesProjects, 2024-01-05:
@baldram That's ok. Let me first make a real basic example as the code I have is a PR in a bigger project.

Also I need to test more than just a couple of minutes.

CelliesProjects, 2024-01-09:
Hi @baldram. I opened #109, showing a possible way to use this library on M5Stack boards.

As the code is in the examples folder, I personally think there is no need to make a specific mention in the README so I skipped that part.

baldram, 2024-01-15:
> As the code is in the examples folder, I personally think there is no need to make a specific mention in the README so I skipped that part.

That's right. The inline code documentation in the example file is enough.