baldram/ESP_VS1053_Library #116: Sound skipping on just one radio.

CBx86, 2025-02-11:
Hey guys!

Using WebRadioDemo;
 i try this online radio: http://5.135.220.118/320 The sound keeps skipping; i thought it might be the bitrate. 
So I changed the bitrate to 128: http://5.135.220.118/128 And Keeps skipping.

Only with this one does this happen.
Try increase/decrease buffer, nothing.
I the board ESP32, CPU uses very low about 20% core 1.
VS1053 v4 using patch.

Some hint?

Many Thanks!

CBx86, 2025-02-11:
It's not really skipping, it looks like it's buffering, but it's not. It looks like something every 500ms.
Sorry for the not-so-precise explanation.

Many Thanks.

Dr-Dawg, 2025-02-12:
maybe chunked encoding?
Have you tried replacing HTTP/1.1 with HTTP/1.0 in the client.print command?

Also see
A way to suppress the glitches and other streaming troubles #52

CBx86, 2025-02-12:
Hey! Thanks for answer. 

The problem is not the glitches, the web radio plays (for 500ms), no sound (for 500ms), (and then plays again for 500ms) as if it were buffering. Or maybe it is..

In reading the issue. And try the RingBufferDemo.cpp.
Strangely, this radio gives a lot of 0 buffering.
`Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 6%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: 0%Buffer: `

Maybe even though the internet is strong, something is coming in slowly.

I will make a code to bench this.
This webradio needs: 124.72 Kbps average.
This ESP32 bench: 2759.45 Kbps.
It's not wonderful but it's ok.

Thanks!

Any other suggestions?

CBx86, 2025-02-12:
More data:
F`ree Heap: 239652
Bytes available: 2960
Free Heap: 237684
Bytes available: 3156
Free Heap: 237140
Bytes available: 1844
Free Heap: 235368
Bytes available: 2488
Free Heap: 239652
Bytes available: 2544
Free Heap: 239652
Bytes available: 2756
Free Heap: 238960
Bytes available: 1444
Free Heap: 239344
Bytes available: 4156
Free Heap: 236628
Bytes available: 2844
Free Heap: 238032
Bytes available: 2964
Free Heap: 237152
Bytes available: 2752
Free Heap: 237976
Bytes available: 4108
Free Heap: 236496
Bytes available: 2796
Free Heap: 237968`

CBx86, 2025-02-12:
Think i need more speed on ESP32 wifi.

I looked for new radios at 320 and found one and it didn't work. Got a ton of logs.

CBx86, 2025-02-14:
After a lot of thought and testing, increased the wifi speed a lot, buffers, etc, nothing worked. Decided to read the datasheet:

[image]

When i use http://5.135.220.118/128 works fine. I made some mistake in not being able to run at 128 in the beginning.

Many Thanks!

baldram, 2025-02-15:
Hey @CBx86 Thanks for the update! Thank you @Dr-Dawg for your support. Glad to hear the solution seems to work. If any issues come up, feel free to reopen. Closing this as resolved.

closed 2025-02-15 by baldram